raspberrypi/rpi-imager #693: Bug in keyboard setup

solskogen, 2023-10-20:
I choose US keyboard in the setting before installation, but after boot it was set to GB.

tdewey-rpi, 2023-10-20:
Thanks for the report, @solskogen. I'll update this issue once we've got a diagnosis.

tdewey-rpi, 2023-10-20:
@maxnet Another if you want it.

maxnet, 2023-10-20:
Which OS image are we talking about?

And it's a normal keyboard? (Not Pi 400, or Raspberry Pi, as that has some keyboard layout override stuff).

solskogen, 2023-10-21:
I used the desktop image (not full) on a Rpi400.

maxnet, 2023-10-21:
>I used the desktop image (not full) on a Rpi400.

Unable to reproduce, when selecting both the checkboxes for "set user" and "set locale", like this:

[image]

[image]

If I only select "Set locale setting", and not user, I do get the piwiz first-usage-wizard thingy, that could make different decisions about keyboard.

solskogen, 2023-10-21:
It happen to me when using the network boot, so it's quite possible that it only happens there. 
[image]
[image]
[image]

maxnet, 2023-10-21:
Ah, that explains things.
Imager version currently offered through netinstall is old, and doesn't support the imager-custom script customization method for RPI OS Bookworm.

Older RPI OS versions and operating systems using cloud-init for customization such as Ubuntu Server do should work.

Should be fixed once the netboot image is updated.

solskogen, 2023-10-23:
Is there any way to see what version of rpi-imager is used in netboot?

maxnet, 2023-10-23:
>Is there any way to see what version of rpi-imager is used in netboot?

As long as it still shows 2022 at https://downloads.raspberrypi.com/net_install/ you have the old one. ;)

There's another issue about the lack of version number. #408

pamoroso, 2023-11-08:
I experienced the same issue on my Raspberry Pi 400 with Italian keyboard when doing a network boot. I'm afraid I don't remember all the details as I burned the microSD card 3 times before realizing what was going on and fixing the installation, but here's an outline of how things went.

First off, I configured every option accessible from the cogs icon of the Imager.

I picked the 64-bit Bookworm "Raspberry Pi OS with desktop and recommended software" image, then set the country to Italy and the language to English. But, when booting into Raspberry Pi OS after the installation, I got wrong password errors every time the password was requested (e.g. to install system updates). This made me realize something was off with the locale. In the end I set the country to Italy and the language to Italian and everything worked fine.

If all password fields in the Imager (e.g. user and Wi-Fi password) allowed showing the password, I would have immediately spotted the language mismatch. You may consider adding such an option whereever it's missing.

Finally, I'm beyond positive I configured Wi-Fi access in the Imager, yet when booting into Raspberry Pi OS after the installation no Wi-Fi was configured in the system.

pamoroso, 2023-11-08:
[I wrote](https://github.com/raspberrypi/rpi-imager/issues/693#issuecomment-1802365642):

> In the end I set the country to Italy and the language to Italian and everything worked fine.

Actually, now that I think about it, setting the country and language wasn't enough to fix the issue.

What I did in the Imager was to enter a user password that contained only lowercase letters and no other letters or symbols. Then, when I booted into the installed system, I immediately changed the password from the shell with the `passwd` command, after typing a few test characters in the shell and checking the terminal displayed the expected letters and symbols.

tdewey-rpi, 2025-08-01:
Marking as closed - Imager was long-since moved to 1.8.5 for network installation, and we're preparing 1.9.7 for the network installer.