raspberrypi/rpi-imager #699: About Telemetry option 1.8.1

baconsplit, 2023-10-21:
telemetry is disabled on default in the settings now,
is that correct?

tdewey-rpi, 2023-10-21:
Thanks for the report, @baconsplit.

That's not what I'm seeing with v1.8.1. Can you give me some more detail on your set-up? Ideally, Host OS, how you installed the Imager, a broad strokes run-through of how you've come to observe this.

baconsplit, 2023-10-21:
Sure, let me also try to replicate. 
I am on Windows 11.
I used version 1.7.5 before but wanted a clean install of rpi with 1.8.1 and thus deleted the cache folder in "%appdata%\..\Local\Raspberry Pi"
Afterwards, I installed and launched the .exe of 1.8.1 and configured the settings before writing on the SD card. Nothing from the previous configuration was set, like device name, SSH login, etc.
But in the new Options tab on the top, Telemetry was deactivated already.

On my try to reproduce just now, I deleted that cache folder again. This time, rp-imager said I already have an saved configuration which I could apply. I said No and delete this configuration, wanting to check the config again. And yes, almost everything like device name, ssh login etc was default again, but telemetry was set to off right away. 

I can provide screenshots of my steps if my broken English does not suffice.

tdewey-rpi, 2023-10-21:
Thanks for the detail, @baconsplit. I'll try and reproduce this on my Windows 11 laptop.

tdewey-rpi, 2023-10-21:
Can you confirm your Windows registry setting for this, by running
`reg query "HKCU\Software\Raspberry Pi\Imager"`

And giving me the result for "telemetry"?

tdewey-rpi, 2023-10-21:
Note that on Windows, just deleting the cache directory isn't strictly enough if you want a completely "clean" installation, you'd want to use the correct uninstaller, which should also remove the registry keys associated with the application.

baconsplit, 2023-10-21:
> Can you confirm your Windows registry setting for this, by running `reg query "HKCU\Software\Raspberry Pi\Imager"`
> 
> And giving me the result for "telemetry"?

Sure. It says
`telemetry REG_SZ false`

> Note that on Windows, just deleting the cache directory isn't strictly enough if you want a completely "clean" installation, you'd want to use the correct uninstaller, which should also remove the registry keys associated with the application.

I figured, since my previous configuration was still found when trying to replicate the issue. I am rather surprised telemetry still stayed at off, even after saying "No, delete configuration" when rpi found a previous config before writing to the SD Card.
Reason for that could be that the registry key was not set to default.

tdewey-rpi, 2023-10-21:
Thanks for the rapid responses, @baconsplit.

Closing this, as this is broadly expected behaviour. The `No, clear settings` button actually clears the OS customisation settings, rather than the application settings. I'll take a note to rework the clearing flow, since it's clearly been too ambiguous in your case.

Hope you find v1.8.1 useful!

closed 2023-10-21 by tdewey-rpi

baconsplit, 2023-10-21:
Also thank you for the quick answers. And interesting, as I expected the default settings on all options when clicking that button. But even better if everything is working as intended. :)

v1.8.1 looks very clean, as for all of rpi. It makes setting up a fresh Pi super easy, thanks for your honest work.