raspberrypi/rpi-imager #706: Support updating wifi from /boot

jedahan, 2023-10-25:
With the migration to networkmanager, we lost the ability to easily update wifi credentials by mounting /boot/ and writing a wpa_supplicant.conf file.

I'd like to bring this back in whatever the most appropriate manner.

Was thinking of just adding `mv /boot/*.nmconnection /etc/NetworkManager/system-connections` to firstrun.sh, but wanted to start a small discussion to make sure this is the best way for the rpi team and community first.

jedahan, 2023-10-25:
The other option I see relates to cloud-init, which supports `updates:` clauses for running "per-boot" instead of first boot, but I am not sure how that integrations works with rpi.

jedahan, 2023-10-25:
See previous discussion on https://github.com/RPi-Distro/raspberrypi-sys-mods/issues/82

XECDesign, 2023-10-25:
I don't think this is the right repo for this. This is all very specific to raspberry pi os and none of this functionality was ever provided by rpi-imager itself.

jedahan, 2023-10-25:
Yeah, it’s a bit hard to find where raspberry pi OS actually defines firstboot.sh - should I open in pi-gen or is there another location?

lurch, 2023-10-25:
See the discussion in https://github.com/raspberrypi/bookworm-feedback/issues/72