raspberrypi/rpi-imager #707: imager ignore the setting i put in (v1.8.1)

Sharonhazan, 2023-10-26:
i use the advance setting, and its seems to ignore all the setting i put in
like hostname and user and password, it run as Admin on win 10

tdewey-rpi, 2023-10-26:
Thanks for the report, @Sharonhazan.

Can you confirm what OS you were attempting to customise?

Sharonhazan, 2023-10-26:
Bullseye lite on rpi 3b

tdewey-rpi, 2023-10-26:
I'm not able to reproduce this failure, up to the point of booting the Raspberry Pi 3B. Imager is writing the customisation as I'd expect.

When you say rpi-imager seems to ignore all the settings, how are you observing the failure - on the Raspberry Pi device itself?

Sharonhazan, 2023-10-26:
> I'm not able to reproduce this failure, up to the point of booting the Raspberry Pi 3B. Imager is writing the customisation as I'd expect.
> 
> When you say rpi-imager seems to ignore all the settings, how are you observing the failure - on the Raspberry Pi device itself?

yes, when i boot the pi (headless) i dont see that he got the hostname i gave him and i don't able to connect to the SSH, even on the deafult pi user and password

Sharonhazan, 2023-10-26:
its seems that there is something in my pc that makes it not to work, tried in my work laptop and it seems to work

tdewey-rpi, 2023-10-26:
@Sharonhazan How did you install v1.8.1? Did you previously have Raspberry Pi Imager installed?

Sharonhazan, 2023-10-26:
from the rpi website
on the pc i think i had, yes

Sharonhazan, 2023-10-26:
HKEY_CURRENT_USER\Software\Raspberry Pi\Imager
    beep    REG_SZ    true
    eject    REG_SZ    false
    telemetry    REG_SZ    true
    x    REG_DWORD    0x400
    y    REG_DWORD    0x25e

HKEY_CURRENT_USER\Software\Raspberry Pi\Imager\caching
HKEY_CURRENT_USER\Software\Raspberry Pi\Imager\imagecustomization

from the pc. which is currently installed

tdewey-rpi, 2023-10-26:
Making sure you remove your password from the output, can you also send me the output from `reg query "HKEY_CURRENT_USER\Software\Raspberry Pi\Imager\imagecustomization"`?

Sharonhazan, 2023-10-26:
The system cannot find the path specified.

Sharonhazan, 2023-10-26:
hostname    REG_SZ    PI-Server
    sshEnabled    REG_SZ    true
    sshUserName    REG_SZ    Sharon
    sshUserPassword    REG_SZ

tdewey-rpi, 2023-10-26:
Thanks, @Sharonhazan. That confirms that Raspberry Pi Imager has at least persisted your settings in the manner I'd expect.

Can you please walk me through your precise button-press sequence in the Imager, so that I can confirm I'm doing the same steps in my testing? Include which buttons you press in any dialog windows that appear, please.

jmfullerton, 2023-10-30:
I've seen the same problem doing customizations for a Raspberry Pi Zero W, Raspberry Pi 3B,  Raspberry Pi 4 and Raspberry Pi 5.  When I boot on the various PIs, it appears that the WiFi hostname hasn't been set.  Every time after I created the first image, when I look at the customizations again  the SSID is still set, but showing the password it is a hash and not what I entered.   I'm running the Imager on Windows 11.   Seems the only solution is to do configuration over a wired ethernet with SSH.  It also took a little bit of time to find out that SSH wasn't configured because it's on a separate tab from all the other network settings.

shanghaix, 2023-11-01:
Having the exact same issue on a Pi4 with version 1.8.1, SSH doesn't even get enabled. Downgraded to 1.7.5 and it works fine again, keeping all the settings.

Sharonhazan, 2023-11-02:
i might managed to fix it?
i deletedt the cache on the AppData\Local\Raspberry Pi\Imager
and reimaged

tdewey-rpi, 2023-11-02:
@Sharonhazan, thanks again for your updated report. Based on your information, I found https://github.com/NixOS/nixpkgs/pull/197010, which in turn sent me to issues reported in other projects where manually deleting the QML cache would seemingly resolve the problem.

Based on this workaround being used in numerous places, I've added a patch to enforce this in Raspberry Pi Imager. This will increase application start time - potentially noticably. Unfortunately, however, I feel the trade off should favour correctness over perceived load time.

closed 2023-11-02 by tdewey-rpi

Sharonhazan, 2023-11-07:
this function already in the imager or it will be in the next version?

Sharonhazan, 2023-11-07:
ok, thanks
i will wait

Technorocker, 2023-12-03:
I too am experiencing this on both the PI zero W and a 3B which is frustrating when Im trying to setup a headless installation and have to question why and whats going on cause it never correctly gets the WIFI settings working. Then I start questioning all the settings and start hunting through config files to see what was and wasnt applied

I just did a reimage for the 3B and reentered all the config settings in imager. luckily I went desktop version(which Id rather not) to see what was going on. on boot it does show the hostname and the username I chose but the network didn't connect. I opened the wpa_supplicant.conf file and it didn't have any WIFI settings in it. just 

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```
my output from reg query "HKEY_CURRENT_USER\Software\Raspberry Pi\Imager\imagecustomization"

```
C:\WINDOWS\system32>reg query "HKEY_CURRENT_USER\Software\Raspberry Pi\Imager\imagecustomization"

HKEY_CURRENT_USER\Software\Raspberry Pi\Imager\imagecustomization
    hostname    REG_SZ    rhasspy1
    keyboardLayout    REG_SZ    us
    sshEnabled    REG_SZ    true
    sshUserName    REG_SZ    user1
    sshUserPassword    REG_SZ    $5$7MJw/YuZOB$FxJpJgN.m/sH.sdCuXqQLFGRNAgIuG/kGrjvM8u.187
    timezone    REG_SZ    America/Toronto
    wifiCountry    REG_SZ    CA
    wifiPassword    REG_SZ    9535a491cd82d4ef87995931c665c74d42ae943ecd50465549cb4320a5bb0409
    wifiSSID    REG_SZ    mywifi
```
I am also using v1.8.1 imager

maxnet, 2023-12-03:
> I opened the wpa_supplicant.conf file and it didn't have any WIFI settings in it. 

That's normal now RPI OS switched to NetworkManager.
Doesn't use wpa_supplicant.conf but sends the configuration to wpa_supplicant through other channels.

Can normally see connection profiles with "nmcli con"
Although I am not sure whether or not it remembers the connection if it never succeeded in connecting at least once...

lonestarmick, 2024-12-27:
still having this same issue , random hash is generated after making bootable sd card

lurch, 2024-12-27:
> still having this same issue , random hash is generated after making bootable sd card

That's deliberate. It's not a "random hash", it's a secure hash of your password, so that your password isn't written to the SD card in plain-text (which could easily be read by anyone).

lonestarmick, 2024-12-27:
Your correct, it's generated a hash , but when trying to ssh or using the username and password, no luck logging in. If I go into the first boot.sh folder and change the ssid password to plain text password. It works then. So for the username should I just create my own sha 256 hash and input that instead ?

lurch, 2024-12-27:
No, no need to create your own hash. Raspberry Pi Imager should "just work", and it has always worked fine all the times that I've used it. Are you 100% sure that you've typed in your SSID and password etc. correctly? If you're able to temporarily connect an HDMI display and USB keyboard to your headless Raspberry Pi, that'll make it much easier for you to figure out what's going wrong.

lonestarmick, 2024-12-27:
Yes , copies the SSID exactly with password , the SSID does have space in it , not sure if that would cause a issue. I am using the pi zero wh, and unable to connect via HDMI , I tried use gadget mode , and use micro USB cable to connect it via Ethernet/network device, with no luck. I have changed the said password to plain text in the first boot sh folder and it was able to connect but then tried the username and password to ssh , I get confirmation on the fingerprint, and when it's time to enter password , denied Everytime. Used default ssh for pi and also the username and password I entered in the pre set options on the pi imager

lurch, 2024-12-27:
Are you using any "special characters" (e.g. `"#@!$£|`) in your password? As a test, could you see if it works when you use only letters and numbers in your password?
I guess if you're getting asked for confirmation of the fingerprint, then that shows that the Pi **is** connected to your local WiFi network, and so the space in the SSID obviously isn't a problem :slightly_smiling_face:

lonestarmick, 2024-12-28:
Ahh good 🤙, yeah it seems the problem is around the ssh password part. I honestly had to put it down for a while. Will try again soon

lonestarmick, 2024-12-31:
> Are you using any "special characters" (e.g. `"#@!$£|`) in your password? As a test, could you see if it works when you use only letters and numbers in your password? I guess if you're getting asked [...]

dude,  i feel so stupid after figuring it out, was 100% the keyboard layout,  . . . . .  ughhhh

PR #713 (2023-11-02): qml: Disable disk cache entirely
This is a compromised change. Essentially, I'm working around a poorly-understood failure mode in Qt5, where the disk cache can occasionally fail to invalidate, resulting in stale compiled units being used.

This will increase application start times in favour of correctness.

Likely to resolve #707