raspberrypi/rpi-imager #708: Ubuntu logo in imager OS choice menu out of date

cassiocassio, 2023-10-26:
[image]

You should find the assets you need to update here: https://design.ubuntu.com/resources

tdewey-rpi, 2023-10-26:
Thanks for the report, @cassiocassio.

Unfortunately, we do not provide this asset. I'd encourage you to contact Canonical directly, and ask them to update the image they're serving. For reference, the icons in the image list are -- with the exception of Raspberry Pi provided OSes and EEPROM files -- provided by the OS vendor.

To assist you with your report, you can point to "https://assets.ubuntu.com/v1/85a9de76-ubuntu-icon.svg", which is taken from our OS image list, served at https://downloads.raspberrypi.org/os_list_imagingutility_v3.json.

Closing, as this isn't something we can correct in the rpi-imager codebase.

cassiocassio, 2023-10-26:
Thanks for your quick response @tdewey-rpi - I am with the Canonical Design team. I'll raise an issue at our end and try and connect the right people. We might have to get you a new link/path or we may be able to update that asset, depending on what else also depends on it.

spl237, 2023-10-26:
@ghollingworth  - it looks as if we may want to get the web team to update the icon URL in the os_list.json file?

tdewey-rpi, 2023-10-26:
> Thanks for your quick response @tdewey-rpi - I am with the Canonical Design team. I'll raise an issue at our end and try and connect the right people. We might have to get you a new link/path or we may be able to update that asset, depending on what else also depends on it.

Ah, in which case: pleasure to (virtually) meet you!

Let me know if there's additional support you need from our side.

@spl237 I don't think that's necessary - I can make the change if @cassiocassio requires a different URL.

META: Reopening the issue to track any changes we may need to make server-side.

tdewey-rpi, 2023-11-01:
@cassiocassio Just checking in - is there anything you require from us to help facilitate your branding update?

tdewey-rpi, 2023-11-20:
Closing - no response in a month, so I'll assume we don't need to provide anything on our side, or you'll contact us directly.

closed 2023-11-20 by tdewey-rpi