raspberrypi/rpi-imager #711: WiFi AP

w0lf29, 2023-10-31:
Add an option to configure WiFi AP at first boot.
Useful for wireless SSH without existing WiFi Network

cillian64, 2023-11-01:
Thanks for the feature request, agreed this would be useful.  But please do not assign issues, we will do that when allocating what we're working on

cillian64, 2023-11-01:
Sorry, I didn't realise the assignees were automatically added by the new issue template!

tdewey-rpi, 2023-11-01:
Apologies, @cillian64. Hoped to mention it next time I saw you. Let's have a chat about workflow offline.

maxnet, 2023-11-01:
Do be aware that offering this functionality (and other advanced network functionality that has been previously rejected such as static IP-addressing) to end-users of varying technical abilities is not without risk.
If someone accidentally chooses the wifi hotspot option instead of wifi client, and enters his normal home network SSID and password, other devices in the home may connect to it, and will not receive Internet...

tdewey-rpi, 2023-11-30:
Thanks for the report, @w0lf29.

I've considered both your suggestion and @maxnet's concerns, and concluded that @maxnet is probably correct on this particular issue. While setting up a SoftAP is an extremely helpful feature, the risk of creating a rogue AP by mistake using Imager is both non-zero and the consequences severe, requiring advanced knowledge of networking to resolve.

As stated elsewhere, my bias is to favour the new and inexperienced users of Raspberry Pi Imager - and I don't believe it's fair to expect a new user to track down and resolve a rogue AP.

I don't have a monopoly on good ideas. I'm happy to accept a counterproposal, but I would expect any counterargument to consider the issues already raised. Closing this for now, but I'll keep an eye on responses.

closed 2023-11-30 by tdewey-rpi

lurch, 2023-11-30:
@w0lf29 If you've not seen it already, you may find https://www.raspberrypi.com/documentation/computers/configuration.html#host-a-wireless-network-on-your-raspberry-pi useful.