raspberrypi/rpi-imager #720: [BUG] Build fail on Debian 12 Bookworm (Raspberry Pi OS host), missing dependencies

alxlve, 2023-11-10:
**Describe the bug**
Build fail if the packages `pkg-config` and `libdrm-dev` are not installed.

**To Reproduce**
Run these commands on a Raspberry Pi 4 with Raspberry Pi OS (Raspbian) and Docker installed.
```

docker pull debian:12.2
docker run -it debian:12.2 /bin/bash
```

```
apt update
apt install --no-install-recommends build-essential devscripts debhelper cmake git libarchive-dev libcurl4-gnutls-dev liblzma-dev qtbase5-dev qtbase5-dev-tools qtdeclarative5-dev libqt5svg5-dev qttools5-dev libgnutls28-dev qml-module-qtquick2 qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtgraphicaleffects

git clone --depth 1 https://github.com/raspberrypi/rpi-imager
cd rpi-imager
git fetch --tags
git checkout v1.8.1

debuild -uc -us
```

**Desktop (please complete the following information):**
 - OS: Debian 12 Bookworm (Docker image)

**Resolution**

Run `apt install pkg-config libdrm-dev` before building `debuild -uc -us`.

lurch, 2023-11-11:
Just out of curiosity, why are you building inside a Docker image rather than just building directly inside Raspberry Pi OS itself?

alxlve, 2023-11-11:
> Just out of curiosity, why are you building inside a Docker image rather than just building directly inside Raspberry Pi OS itself?

I don't have to keep track of installed apt dependencies for the build. Some projects have also such build that they pull a lot of files and keep them installed (such as Go projects or Python ones, depending on how the build was described).

Running in Docker let me start from a fresh and clean system and destroy it in seconds when the build is over. There is no impact on the host system too and it is convenient to me. I do not risk my host when building new projects and unfortunately sometimes poor designed build steps. `rpi-imager` build is clean, I like it.

I can't only uninstall the list of apt dependencies provided for the build, if so I would have to check manually that each dependency was not installed before, because they may have been already used by some other tools on the system. That's a struggle and running in Docker give me the opportunity to not wonder myself any question, I just follow the instructions from the project and then destroy the container.

maxnet, 2023-11-11:
Suggest you check out default branch (qml) instead of version tag.
Was fixed a while ago in 457f2fdbedd86cd29276a883e223765306d03311

tdewey-rpi, 2023-11-13:
Known issue in v1.8.1, fix will be included in v1.8.2 tag.

closed 2023-11-13 by tdewey-rpi