raspberrypi/rpi-imager #721: [BUG] SSH credentials not generated on rasberry pi 4 for Mainsail OS

lowyoujian, 2023-11-11:
**Describe the bug**
A clear and concise description of what the bug is.
Unable to ssh into raspberry pi after enabling ssh on raspberry pi imager with username and password.

**To Reproduce**
Include steps to allow us to reproduce the error. Include your host OS,

1. Selected Raspberry pi 4 for device,
2. Selected Mainsail for OS 
3. Popup appears to edit configurations
4. Entered my localhost, username and password, go to next tab and enabled ssh.
5. Write.

**Expected behavior**
A clear and concise description of what you expected to happen.
ssh file should be generated with my username and password, but only found default pi user with random hashed password in userconf.txt

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. macOS 14, Windows 11, Ubuntu Server 22.04]
OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045
and
MacOs Ventura 13.0.1

**The name of the OS you are trying to write**
Mainsail 1.2.1

**Are you using OS Customisation?**
No
**Additional context**
I Actually downloaded 1.7.5 release version and it worked perfectly.

Sharonhazan, 2023-11-12:
try clear the cache on AppData\Local\Raspberry Pi\Imager, it **might** help

tdewey-rpi, 2023-11-13:
Thanks for the report, @lowyoujian. We're about to release v1.8.2, which may help resolve this problem.

BrandonRichards-Valore, 2023-12-03:
This appears to be broken still. I've tried 1.8.1 and 1.8.3 on Windows 11, I am installing PI OS lite 64bit (I have the 8gb version), and set the user name and password to my name all lower case just to rule out weird character encoding as some posts on the internet seem to suggest. I also tried clearing the AppData cache. I tried, both password auth and public key SSH setting from the imager. It doesnt matter what I do, I an unable to SSH into a newly imaged Pi 4B.

coreConvention, 2023-12-03:
This appears to be broken still. I've tried 1.8.1 and 1.8.3 on Windows 11, I am installing PI OS lite 64bit (I have the 8gb version), and set the user name and password to my name all lower case just to rule out weird character encoding as some posts on the internet seem to suggest. I also tried clearing the AppData cache. I tried, both password auth and public key SSH setting from the imager. It doesnt matter what I do, I an unable to SSH into a newly imaged Pi 4B.

lurch, 2023-12-03:
> I am installing PI OS lite 64bit...

Why are you commenting on an issue about Mainsail OS? You might find some helpful info in #531 instead

maxnet, 2023-12-03:
>It doesnt matter what I do, I an unable to SSH into a newly imaged Pi 4B.

With what exact error message?

As you would also be "unable to ssh", if for example you were using the wrong wifi settings or country code (of using 5 ghz wifi).

And you do are under using "ssh username@name-of-pi.local" from the windows command prompt?
(Instead of 3rd party ssh clients like putty, which may or may not support the right algorithms for recent RPI OS versions)

coreConvention, 2023-12-03:
> > I am installing PI OS lite 64bit...
> 
> Why are you commenting on an issue about Mainsail OS? You might find some helpful info in #531 instead.

Haha my bad I didn't read the title just read the post and it looked like the exact same issue I have been having.

coreConvention, 2023-12-03:
> > It doesnt matter what I do, I an unable to SSH into a newly imaged Pi 4B.
> 
> With what exact error message?
> 
> As you would also be "unable to ssh", if for example you were using wifi, and had the wrong wifi settings, country code (if using 5 ghz wifi or channel 13), or a wifi AP that only speaks wpa3 instead of wpa2. May want to verify that you do can ping the Pi.
> 
> And you do are typing "ssh [username@name-of-pi.local](mailto:username@name-of-pi.local)" in the windows command prompt when using pub key authentication? (Instead of 3rd party ssh clients like putty, which may or may not support the right algorithms for recent RPI OS versions, and expect the key file at different location)

I have verified connectivity via my UniFi controller and I can ping it over wifi and yes I am using the user@name.local. I have tried to connect via putty as well as the windows native ssh client.

Anyway, it was not my intention to crap up this thread. Just thought the issue seemed similar.

derekste, 2023-12-04:
1.8.1 installer is extremely buggy on Mac - none of the SSH options work across any OS. 1.8.3 fixed my issue.

tdewey-rpi, 2023-12-13:
@lowyoujian Are you able to reproduce this issue on 1.8.3?

tdewey-rpi, 2023-12-15:
I believe this may be related to #531, which should be resolved imminently.

asha77, 2023-12-15:
I check this issue with mainsail os:
- issue reproduced in 1.8.1.
- issue not reproduced in 1.8.3 (after upgrade) - i successfull get access to os via ssh using credentials i setup in configuration

closed 2024-01-22 by tdewey-rpi