raspberrypi/rpi-imager #724: [BUG]SSH Setup and SSH-KEYGEN grayed out

Grip3n, 2023-11-15:
**Describe the bug**
Setting up SSH via the Imager and using 2023-10-10-raspios-bookworm-armhf-lite.img I get the following error
Server refused our key
No supported authentication methods available (Sever send: publickey)

But when using the Imager with 2023-05-03-raspios-bullseye-armhf-lite.img.xz it works with no issues.

Another issue I noticed was that the Run SSH-keygen gets grayed out after running it once.
Cannot create a new key for each write.

**To Reproduce**
SSH setup:
Downloaded Raspberry Pi Imager v1.8.1
Create key using PuTTYgen
Raspberry Pi Device: Raspberry Pi4
Operating System:  Use Custom
First test: 2023-05-03-raspios-bullseye-armhf-lite.img.xz
Second test:  2023-10-10-raspios-bookworm-armhf-lite.img
Use OS Customization --> Edit settings (see Screenshots section) --> The settings does not change between the two writes

Run SSH-KEYGEN:
Select Enable SSH
Select "Allow publick key authentication only"
Click "RUN SSH-KEYGEN"
Try and run it again
Clear SSH Key field
Try and run it again
(Notice in Screenshot it's grayed out)

**Expected behavior**
SSH should work as in Bullseye setup
Should be aible to create new keys as needed ?

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10

**The name of the OS you are trying to write**
2023-10-10-raspios-bookworm-armhf-lite.img

**Are you using OS Customisation?**
Yes, See Screenshots section

**Additional context**
It may just be a PICNIC issue ?
But i have tried this now with 2 SD cards and fresh writes of both images and results are the same.
works with Bullseye but not with Bookworm

tdewey-rpi, 2023-11-15:
Thanks for the report, @Grip3n.

I see two issues in this report:

1. `RUN SSH-KEYGEN` is greyed out. This is because you've not installed the Windows OpenSSH client, and are instead using PuTTYs generator. The expected path of the binary is `windir+"\\SysNative\\OpenSSH\\ssh-keygen.exe"`, and without it you will not be able to run ssh-keygen within Imager.
2. You failed to log in with public key authentication. You appear to have misconfigured your client, because the negotiation of authentication mechanism is failing. `No supported authentication methods available (Sever send: publickey)` means that your client cannot perform publickey authentication.

In short, neither of these are issues with Raspberry Pi imager. To resolve (1), you could consider using the Microsoft-provided guidance at (https://learn.microsoft.com/en-gb/windows-server/administration/openssh/openssh_install_firstuse?tabs=gui). To resolve (2), you'll need to consider your SSH client documentation - it's possible that you've incorrectly configured your SSH agent or similar, but support in this isn't something I can provide.

Grip3n, 2023-11-15:
Good day @tdewey-rpi

1.  Please note that OpenSSH is infact on my system if I naviate to %systemdrive%\Windows\System32\openssh
is see five ssh files including ssh-keygen.exe
I can execute the file as standalone.

2. I see your point in with your statement "You appear to have misconfigured your client"  and i will raise you the question,
Leave the SSH client AS IS that works with Bullseye with no issues what so ever
and only change the Image to Bookworm within the Imager and SSH does not work.

maxnet, 2023-11-15:
>windir+"\\SysNative\\OpenSSH\\ssh-keygen.exe"

Sysnative is an alias for \windows\system32 on your windows system.
(Special construct. System32 is hidden to 32-bit applications on 64-bit windows)

- in the openssh world ssh keys are normally per person, not per host. You do can paste the public keys of multiple people in there.

- no need to hide your public key in screenshots. You can share them with the world. Your corresponding private key is NOT shown.

- suggest you give the normal openssh client a try. Just open Windows command promt and enter "ssh pi4lab.local"

Grip3n, 2023-11-15:
Good day @maxnet
I am not so much worried about the SSH-Kegen creation part as i use PuTTYgen

what still bugs me is the following,

Setting up PuTTY  
Connection --> SSH--> Auth --> Private key file for authentication --> points to .ppk file

I created a new Private and Public key pair for testing
As can see by the Screenshots
The only thing that has changed is the Operating System and the Public key.

PuTTY, Bullseye and Imanger 1.8.1 work from the get go.
PuTTY, Bookworm and Imanger 1.8.1 does not work

Connection --> SSH--> Auth --> Private key file for authentication --> points to .ppk file has not changed 

[image]
[image]

maxnet, 2023-11-15:
Recall there are changes in what exact SSH ciphers are supported in recent Debian versions, and by extension in RPI OS.
Could be that your Putty version does not use those.

May be able to dumb security settings down in sshd_config.
But that is not somethijg Imager will do for you.

tdewey-rpi, 2023-11-15:
Reopening, as this is still getting engagement.

sysnative appears to be nonfunctional on Windows 11, and it isn't clear how to resolve access of System32 at this point - I've tried a few options, all fail. I recommend we remove the ssh-keygen option on Windows until we have a solution.

@Grip3n Try using an ed25519 key.

maxnet, 2023-11-15:
>sysnative appears to be nonfunctional on Windows 11

Definitely works on Windows 11 for me.
Again: only if you are a 32-bit application.
You won't find \windows\sysnative in your 64-bit Windows explorer.

For background, see: https://www.samlogic.net/articles/sysnative-folder-64-bit-windows.htm

>@Grip3n Try using an ed25519 key.

Recall it is the signing algorithm used during public key authentication.
There is original ssh-rsa which uses insecure SHA1, and newer ssh-rsa-sha256.
OpenSSH client can use the new algorithm even with old key.
No idea what the Putty version used is or is not capable of.

Grip3n, 2023-11-15:
@maxnet 
I found the following [https://michael-prokop.at/blog/2023/06/11/what-to-expect-from-debian-bookworm-newinbookworm/](url)

"One important change you might wanna be aware of is that as of [OpenSSH v8.8](https://www.openssh.com/txt/release-8.8), RSA signatures using the SHA-1 hash algorithm got disabled by default, but RSA/SHA-256/512 AKA RSA-SHA2 gets used instead."

And it seems PuTTY v0.79 got it setup by defualt

[image]

@tdewey-rpi 
I will give it a try 
Still do not know why RSA

tdewey-rpi, 2023-11-30:
Did you have any success with the alternative key type, @Grip3n? We're now 2 weeks since the last update, and this would usually meet my criteria for closing as stale, however I'm aware we never asked you for an update.

tdewey-rpi, 2024-02-08:
Closing this issue. OP has obtained success by using a newer key type, and the original report appears to have been answered by @maxnet.

Changes to handling SSH identities will be planned for 2.0.

bckelley, 2025-08-31:
this is still an issue why caan't @raspberrypi fix there crap! can't set public-keys on either bookworm or bullseye and I'm on linux so you can't say openssh isn't install because it's a core package!