raspberrypi/rpi-imager #726: Bookworm not available

heis2201, 2023-11-16:
**Describe the bug**
I do not have to option to select bookworm. All I can select from is bullseye ("Legacy").

**To Reproduce**
Host os is Fedora 38 up-to-date. both rpm and flatpak show the same behaviour

**Expected behavior**
Raspberry Pi OS 12 "bookworm" available

**Screenshots**
[image]

**The name of the OS you are trying to write**
Raspberry Pi OS 12 "bookworm" LITE

**Additional context**
I saw bookworm (with release day 2023-10-10), then rpi-imager couldn't write the image complaining about permissions on the SD card (probably I still had some partition on the SD card mounted). On the second attempt book worm does not show up as an option

heis2201, 2023-11-16:
related: https://forums.raspberrypi.com/viewtopic.php?t=358721

lurch, 2023-11-16:
Duplicate of #690 ?

heis2201, 2023-11-16:
> Duplicate of #690 ?

indeed.

closed 2023-11-16 by heis2201