raspberrypi/rpi-imager #730: [FEATURE]: Understand the reason bookworms isn't recommended on Pi3

heis2201, 2023-11-17:
In https://github.com/raspberrypi/rpi-imager/issues/690#issuecomment-1772400836 I read that rpi-imager does not recommend bookworm on a Pi3.

I'd like to understand why given the fact that raspberrypi.com does recommend it, see screenshot below.
I am running bookworm aarch64 (64 bit) LITE (no desktop) on a Pi3 with some docker containers and it's working flawlessly.

[image]

Bulwagga, 2023-12-09:
I installed bookworm lite 64 bit to a raspberry pi 3B using rpi-imager from ubuntu 13.10. All worked fine except the pre-configured wireless setup. There may be other gotchas.

lurch, 2023-12-09:
> I installed bookworm lite ... All worked fine except the pre-configured wireless setup.

See #749

tdewey-rpi, 2023-12-15:
As I have commented elsewhere, there is an important distinction between 'compatible with' and 'recommended for new users'. Currently, Bookworm is not 'recommended for new users' of pre-Raspberry Pi 4 devices. There are technical concerns around the graphical user interface and behaviours that appear on lower-memory devices that must be resolved before the recommendation is offered.

Again, as I've said elsewhere: that does _not_ mean you can't try and run it, but you may see unexpected behaviours that would not be replicated on Raspberry Pi 4 and later.

I appreciate this isn't the answer that you may be seeing, @heis2201, however it is the one I consider fairest to new users today.

Closing this issue, as it is not an issue with rpi-imager, but with the OS list.

closed 2023-12-15 by tdewey-rpi

heis2201, 2023-12-15:
Your post answers my question perfectly fine. Thank you!
im running bookworm without DE, so that’s maybe why I haven’t experienced any problem.

henryptung, 2024-03-30:
@tdewey-rpi To make this a little more explicit, is there a citation regarding this remark:

>  Currently, Bookworm is not 'recommended for new users' of pre-Raspberry Pi 4 devices. There are technical concerns around the graphical user interface and behaviours that appear on lower-memory devices that must be resolved before the recommendation is offered.

Just so people can analyze the concerns for themselves and make an informed decision?