raspberrypi/rpi-imager #731: adding option to make image for the sd card

Sharonhazan, 2023-11-17:
i want sometimes to create image and distributed it, it will nice to have the option from the imager

tdewey-rpi, 2023-11-20:
Hi @Sharonhazan, thanks for the report.

It's an interesting idea - and one that we may explore in a separate tool. Today, you can use `pi-gen` to create a customised Raspberry Pi OS, which you can then flash via the 'Use Custom' option in Imager, or distribute after filing the request to add an OS with us.

Unfortunately, however, this isn't something we would look to add for Imager - though I'm happy to entertain a reasoned argument with a PR, which would be graded for it's trade-offs against the non-technical user experience.

Closing as 'Won't Address'.

closed 2023-11-20 by tdewey-rpi

cmitu, 2023-11-22:
@tdewey-rpi I think the request was more of 'how do I backup to an .img file from a sdcard contents' rather than 'how can I create a customized RPI OS'. For instance, `win32diskimager` [1] has this option and was a recommended tool for writing the images before RPI Imager came along.

[1] https://sourceforge.net/projects/win32diskimager/

lurch, 2023-11-22:
> I think the request was more of _backup to an .img file from the sdcard_

That has already been requested and closed multiple times before: https://github.com/raspberrypi/rpi-imager/issues?q=is%3Aissue+backup+is%3Aclosed

tdewey-rpi, 2023-11-22:
> @tdewey-rpi I think the request was more of _backup to an .img file from the sdcard_ rather than _how can I create a customized RPI OS_. For instance, `win32diskimager` [1] has this option and was a recommended tool for writing the images before RPI Imager came along.
> 
> [1] https://sourceforge.net/projects/win32diskimager/

As pointed out by @lurch, that isn't necessarily a new request. as I have stated elsewhere, in terms of determining what to bring to Imager I use the experience of the newest or least technically capable users to guide me - and I struggle to see how we could meaningfully add SD card cloning to Imager without compromising that experience, or even that the use-case is particularly prevalent among less technically capable users.

As ever, I don't consider my position fixed - as anyone who'd previously worked with me may recall, I don't have a monopoly on good ideas. That said, I have put some considerable thought into adjacent features for Imager, and would expect a successful proposal to demonstrate a corresponding amount of thought having been put into it.

Maintaining as 'Won't Address'.

lurch, 2023-11-22:
> I use the experience of the newest or least technically capable users to guide me - and I struggle to see how we could meaningfully add SD card cloning to Imager without compromising that experience

For the benefit of people thinking "That's easy, surely you just read the SD card to an image file"; one of the (several) pitfalls here is that an "8GB" SD card from one brand might actually be 7.9GB, whilst an "8GB" card from another brand might actually be 7.8GB; and less technically capable users will get confused when they're not able to write their 7.9GB backup onto a different 7.8GB SD card. (which is a problem that frequently crops up at https://github.com/balena-io/etcher/issues )