raspberrypi/rpi-imager #735: [BUG] Can't read text in customization menu. Font seems to be white on a white background

MostHated, 2023-11-19:
**Describe the bug**
A clear and concise description of what the bug is.
Can't read text in the customization menu. The font seems to be white on a white background

**To Reproduce**
Include steps to allow us to reproduce the error. Include your host OS,
I installed the 1.8.1 deb which was linked from a site, but then I saw 1.8.3 was available, but the issue occurs the same in both versions.

**Expected behavior**
A clear and concise description of what you expected to happen.
Can't read text. The font seems to be white on a white background

**Screenshots**
If applicable, add screenshots to help explain your problem.
[](https://img.instance.id/hOSxCclqq4Px/direct)

[rpi_imager.webm](https://github.com/raspberrypi/rpi-imager/assets/3442753/08ad0ef8-cb88-4eb8-b34b-e30e2b8e09c9)

**Desktop (please complete the following information):**
 - OS: [e.g. macOS 14, Windows 11, Ubuntu Server 22.04]
Ubuntu 22.04

**The name of the OS you are trying to write**
The current default recommended one, which I believe was debian bookworm?

**Are you using OS Customisation?**
Yes, that is where the main issue occurs

**Additional context**
Add any other context about the problem here.
Not sure if it's theme related, or something? I don't believe I have anything special going on, on the host OS in which I am running the imager. It is all pretty stock in order to stay in spec for Unity3d support.

lurch, 2023-11-20:
I looks like you've enabled a "dark" theme (because of the white text on a grey background on the tab-labels and text-boxes) ? But it looks like some parts of RPi Imager _are_ using the theme-colours, but some parts are ignoring the theme-colours?

maxnet, 2023-11-20:
Any difference if you start a terminal, and enter:

```
LIBGL_ALWAYS_SOFTWARE=1 rpi-imager
```

tdewey-rpi, 2023-11-20:
Took a quick look at the OptionsPopup code - and sure enough, we're not actually enforcing a colour beyond the items that are still visible in the video.

MostHated, 2023-11-20:
Glad to hear it was relatively easy to locate the issue. I had to try and scour a number of websites trying to find out what the text on those tabs said, which wasn't as easy as it sounds. At least for the last page. Most sites only showed the first and second pages.

@maxnet I gave that a try just to see if it made a difference, though it did not seem to have any effect.

maxnet, 2023-11-20:
If it's the theme than you can probably use "QT_QUICK_CONTROLS_MATERIAL_THEME=Light rpi-imager" as a workaround.
But it is uncommon for Ubuntu to set that. There normal light/dark setting only changes the color of title bar.

lurch, 2023-11-21:
> I had to try and scour a number of websites trying to find out what the text on those tabs said, which wasn't as easy as it sounds.

There's a big bunch of screenshots in the [documentation](https://www.raspberrypi.com/documentation/computers/getting-started.html#raspberry-pi-imager). @tdewey-rpi Do you think it's worth linking to there from this project's README.md ?

tdewey-rpi, 2023-11-21:
@MostHated Could you please try and load `rpi-imager` with a theme override? I'd like to understand if this is an interaction with the Qt5 default theme and your Dark Mode theme:

Try:

`rpi-imager -style gtk2`

MostHated, 2023-11-21:
Sure thing. Looks about the same.

[image]

tdewey-rpi, 2023-11-24:
Thanks for the test, @MostHated.

It does indeed look like the Qt5 default themes are not interacting well with your OS theme - with some controls in your installation getting the theme assets, but others not - most notably, the default QtQuick.Window control we're using appears to have ignored the request for a dark base colour - which the parent window _did_ get.

I'll begin investigations on the difference between how we're setting up these Windows, which might get me to the solution. In the meantime, any details you can provide me on how to configure a desktop to match your theme & OS combination would be much appreciated - as I don't want to annoy you with an endless stream of test builds.

maxnet, 2023-11-24:
>It does indeed look like the Qt5 default themes are not interacting well with your OS theme

It's Qt5's own black material design theme that you are seeing.
Which you can also play with yourself by setting the right environment variable:

[image]

The question do is why that is being used.
Just setting Ubuntu OS theme to "black" is not sufficient to cause the Qt theme to be used...

MostHated, 2023-11-25:
It looks like I had set a few env variables prior related to QT apps. Though, this is the first time I have ever seen it have any sort of negative effect, and as such, I don't think about it often, and forget I ever did it.

(Redacted a few personal items)

[image]

tdewey-rpi, 2023-11-26:
Thanks for the update, @MostHated 

Sure enough - setting QT_QPA_PLATFORMTHEME="qt5ct" means that we're no longer using the default Qt themes at all - instead, we're using the configuration specified in the Qt5 configuration tool (as you've shown us). This is a bit of a conundrum:

- Qt5Ct appears to override any attempt by the application to set an app palette - which means a workaround is going to be very involved (at least). There's a patch outstanding to let applications control their own palette: https://sourceforge.net/p/qt5ct/tickets/97/
- I don't generally entertain user-specific configuration problems once we've found them - support for rpi-imager is best-effort and user-specific configuration issues rarely affect more than one user, making them a less effective use of time.

That said, I'm not going to close this - because although there's clearly some missing override (we appear to have picked up your change to the foreground font colour, but not the base window colour).

We'll need to decide what we want Imager to do in these situations - do we just completely ignore user customisation and try to work around the palette override (probably the simplest path), or do we try to map the Options dialog to the Window customisation that seems to work on the main window.