raspberrypi/rpi-imager #739: [BUG] Tab-header in options dialog goes crazy if you enter a long username

lurch, 2023-11-21:
**Describe the bug**
In the "OS Customisation" window, if you enter a _really_ long username, then the tab-buttons at the top of the window get super-wide and the right-most tab ends up disappearing.

**To Reproduce**
Select an Image and Storage Device, press Next, select "Edit Settings", enter a really long username.

**Expected behaviour**
All three tabs should remain visible in the options window.

**Screenshots**
First screenshot shows what it looks like with a short username `abc` :
[image]

Second screenshot shows what it looks like with a long username `abcdefghijklmnopqrstuvwxyzabcdefghijklmnopqrstuvwxyz` :
[image]

**Desktop (please complete the following information):**
 - OS: Raspberry Pi OS Bookworm 64-bit, running on a Raspberry Pi 5

**The name of the OS you are trying to write**
N/A

**Are you using OS Customisation?**
Yes, that's where the bug happens :slightly_smiling_face: 

**Additional context**
It's obviously unlikely that somebody would enter a **real** username that long, which makes this low-priority, but it's still a UI-bug which is probably worth fixing.

tdewey-rpi, 2023-11-21:
Confirmed it appears on Windows if you use a >51 character username.

tdewey-rpi, 2023-11-21:
@lurch has correctly identified this is seemingly caused by the "Set authorized_keys" label on the 'Services' page, which should probably be width-constrained.

tdewey-rpi, 2023-11-21:
This opens a wider question: are we correctly limiting the size of these fields to match established Linux distro norms? Eg, do we need to cap the Username field at 32 characters to be compatible with a range of downstream OSes that may be customisable?

tdewey-rpi, 2023-11-21:
Certainly for the username, I don't see a solid argument to go beyond 31 characters, as systemd will not allow this: https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html

We may need to constrain other labels too.

tdewey-rpi, 2023-11-21:
The hostname must additionally be constrained to 253 characters, per https://man7.org/linux/man-pages/man7/hostname.7.html

PR #741 (2023-11-21): qml: Impose limits on usernames, hostnames
Resolves #739 by constaining the size of the username field to match the limit imposed by systemd, and the hostname to the limit specified on man7.
closed 2023-11-21 by tdewey-rpi