raspberrypi/rpi-imager #743: [BUG] Error writing file to disk

tunm1228, 2023-11-24:
[image]

[image]

tdewey-rpi, 2023-11-24:
Hi @tunm1228, thanks for the report.

Unfortunately you've not included enough detail for me to begin investigations in earnest - I can see you're on macOS, for example, but not which version. The template that you deleted while putting the screenshot in place includes those fields in order to allow me to work out how I could go about reproducing the issue, and give me some direction in terms of where to look.

One thing I can immediately recommend is that you upgrade to v1.8.3, which includes critical correctness fixes.

Otherwise, if you cannot provide additional information within the next 2 weeks (by 5PM GMT, December 8th), then I'll be forced to close this issue as 'Cannot reproduce'.

audas, 2023-12-29:
I am having the same issue on windows 11.  Raspberry Pi 4 / Raspberry Pi OS (64 Bit full)

RPImager v1.8.4 
(Have checked to ensure I have access to folder writes in system)

The writer will also write - get to 70-80% then claim "Error reading from storage. SD card may be broken"

I have tried two separate SD cards. 

The passwords are being obfuscated into gobbledy gook after each write too - So I am unable to write multiple cards as each time I must re-enter this information.

I have many raspberry pi and I am unable to install new cards in them as they require wifi and its simply not working. 

Is there some way of rolling back to a previous installation as this release is simply not ready at all for production and should not have been released.

lurch, 2023-12-30:
Well either you're unlucky that both of your SD cards are faulty (were they purchased from a reputable seller? Unfortunately "fake" SD cards are quite common), or maybe your SD card-reader is faulty? Other users have anecdotally reported that sometimes using the USB ports on the back of your PC can be more reliable than the USB ports in the front of your case.

> So I am unable to write multiple cards as each time I must re-enter this information. This is very frustrating when the wifi password is having to be re-written over and over.

That's not true. Even though the password looks like "gobbledy gook" it's actually saved in a hashed format that Raspberry Pi OS is able to use. So no need to keep re-entering the password each time, the "saved" value from the previous time will work just fine.

audas, 2023-12-30:
Yes cards are all fine, major retail chain. USB ports are fine too. I have 30+ pi's plus many other MCU and this release is first time something like this has come up 
The previous release kept the password as there is no way for abuser to KNOW the password is the same as visually they change in the input fields and this is a major change. Have not seen this usx before. Not very functional

lurch, 2023-12-30:
> Yes cards are all fine, major retail chain. USB ports are fine too. I have 30+ pi's plus many other MCU and this release is first time something like this has come up

Just as a point of comparison, it'd be interesting to hear if [Etcher](https://etcher.balena.io/) is more reliable with writing the same image to the same cards (using the same computer and card-reader), or if that suffers from similar problems. Make sure you leave verification enabled.

And the passwords get hashed so that if somebody else happens to use your computer while your back is turned (or manages to steal one of your SD cards), they can't see your passwords written down in plain-text.

tdewey-rpi, 2024-02-07:
Closing as stale - the OP hasn't responded since the initial report, follow up commenters have not followed up in over a month.

closed 2024-02-07 by tdewey-rpi

audas, 2024-02-07:
> Just as a point of comparison, it'd be interesting to hear if [Etcher](https://etcher.balena.io/) is more reliable with writing the same image to the same cards (using the same computer and [...]

I think hashing the passwords is a massive change from the norm in user experience. Nothing else does this as far as I know, or at least it is VERY small cohort.

Its incredibly frustrating to re-write the password every time as there is no way to ensure we are using the correct one. So it MUST be re-written rather than hoping its correct.

All other systems will maintain password access on your OWN computer.

As far as this issue is concerned I have been away for 4 weeks but will test again over the next few days. The cards eventually wrote however only with Lite versions (No head, no desktop).

ghiathkamel, 2024-02-08:
i have same issue verifying error every time after installing the OS on SD card
[image]

RefriZaddo, 2024-02-28:
I'm having this same issue in Windows 11 - Pi Imager 1.8.5 while trying to flash Retropie 4.8 in a 128GB SanDisk Ultra microSD, I tried Etcher, but it doesn't work either.

lurch, 2024-02-29:
@ghiathkamel This issue has already been closed. Please file a new issue, providing all the details asked for in the issue-template.

Kblasz, 2024-07-16:
In case anyone else finds this helpful, I have a bunch of USB 3.0 SD card readers and usb dongles, they all have this exact issue, all i did was unplug them from the USB3.0 (blue) port and put them in the USB 2.0 (black) port. Transfer was much slower but it was able to get to the end.