raspberrypi/rpi-imager #744: [BUG] Mac binary codesign signature expired

MelanieT, 2023-11-26:
**Describe the bug**
The code signature is expired and the program is killed with SIGKILL on MacOS

**To Reproduce**
Download imager 1.8.1 to any version of mac os. Start the program and observe it immediately closes. Open MacOS "Console" utility. In section "crashes", observe an outdated code signature being logged.

**Expected behaviour**
The program should start

**Desktop (please complete the following information):**
Any MacOS, any CPU

**The name of the OS you are trying to write**
N/A

**Are you using OS Customisation?**
N/A

**Additional context**
With codesign --remove-signature rpi-imager, the program can then be started if gatekeeper is set to allow unsigned code to run.

maxnet, 2023-11-26:
Got a screenshot?

And are you sure it is the system killing the application, and not some kind of extra security software you or your employer put on your machine? 

Application was properly notarized. So Apple timestamped and countersigned the thing.

```
spctl -a -v -t install /Applications/Raspberry\ Pi\ Imager.app
/Applications/Raspberry Pi Imager.app: accepted
source=Notarized Developer ID
```

MelanieT, 2023-11-26:
[image]

Happy to oblige.

maxnet, 2023-11-26:
Unable to reproduce.

And I don't see anything indicating there is an expired certificate in your screenshot.
Which is unlikely (besides the file being timestamped by Apple, my certificate isn't due to expire until 2025).
More likely that the file is corrupt/truncated instead.

You did download it from our website, straight to the same machine you are trying to run it on?

MelanieT, 2023-11-26:
Yes. Downloaded your DMG, installed and that is the result. I'm on MacOS Monterey and screenshot says the code signature is invalid at the very end. Maybe the MacOS version has ssomething to do with it?

tdewey-rpi, 2023-11-27:
Thanks for the report, @MelanieT.

@maxnet We've seen a similar collection of issues reported elsewhere - mostly through Social Media channels. Unfortunately, none of those leads followed up. It'd be good to try an understand what about tthe way we're signing Imager requires 13+.

maxnet, 2023-11-27:
>It'd be good to try an understand what about tthe way we're signing Imager requires 13+.

Well, that's the first thing that would require research.
Is it indeed not working on any OSX subrelease before 13?

Note that the TS is using 12.6.3
That is not the latest Monterery subrelease either.
There have been 8 updates since, according to: https://support.apple.com/en-us/106339
And Apple refuses to indicate what changed in those other than that they have "important security fixes"......

Can someone with full access to the rpi-imager-stats telemetry thing figure out what MacOS versions do are able to run Imager?
And if there are older OSX versions success was that with Imager 1.8.* or only with earlier versions?

tdewey-rpi, 2023-12-15:
Sorry @maxnet, took a while to get back to this one.

In practice, the vast majority of macOS users are using 13+, but we see a long tail of users using older versions - all the way down to 10.15.

I've only been able to reproduce this by building on a machine that didn't have an Apple Developer Account signed in - never from an official release.

maxnet, 2023-12-15:
>In practice, the vast majority of macOS users are using 13+, but we see a long tail of users using older versions - all the 
>way down to 10.15.

Ok, so it is not that all users of older versions that are affected.
Would also be interesting to know if upgrading to a newer version does solve it for some, or if the issue persists.

Also surprises me that it only happens after opening the .app
Thought the .dmg file itself is also signed, and would expect OSX to bark if there was something with the signature.