raspberrypi/rpi-imager #746: [BUG] New Icon Doesn't Match Apple Design Guidelines

Logicer16, 2023-11-30:
**Describe the bug**
Following up on #695, the new icon is too large and has a smaller corner radius than what [Apple's Human Interface Guidelines](https://developer.apple.com/design/human-interface-guidelines/app-icons#macOS). This is contrary to the goal stated in https://github.com/raspberrypi/rpi-imager/issues/695#issuecomment-1773740128:
> the desire was [an] icon ... that also complied with the Apple HI guidance.

**To Reproduce**
Include steps to allow us to reproduce the error. Include your host OS, and the version of Raspberry Pi Imager.
1. View the imager's icon next to icons of other Mac apps.

**Expected behaviour**
The new icon to meet the goal of conforming to the Apple Human Interface Guidelines.

**Screenshots or video**
Comparing it with other icons in the dock makes the discrepancy quite striking
[image]
Placing the icon against Apple's provided template shows exactly by how much the new icon is off.
[image]
The micro SD card is fine as is though as it is a tool external from the main icon, which is permitted by the guidelines.

**Desktop (please complete the following information):**
 - OS: macOS 12.7.2

**The name of the OS you are trying to write**
N/A

**Are you using OS Customisation?**
N/A

**Additional context**
N/A

tdewey-rpi, 2023-11-30:
Thanks for the report, @Logicer16.

@jackbenwillis Have I mangled your asset in the merge?

tdewey-rpi, 2023-11-30:
Also, thank you @Logicer16 for taking the time to assemble a visual aide. This is greatly appreciated when reporting UI concerns in particular.

tdewey-rpi, 2023-12-13:
Thanks for the catch, @Logicer16. We'll look to revise the asset, but this is unlikely to happen in 2023.

elsiehupp, 2024-02-29:
I [mentioned](https://github.com/raspberrypi/rpi-imager/issues/397#issuecomment-1584626517) this on #397:

> I have made Big-Sur-style icons for other projects, and most recently I merged this template over at the Nextcloud Desktop repository:
> 
> [image]
> 
> [Nextcloud-macOS-icon.svg](https://github.com/nextcloud/desktop/blob/04d16b7824e633b66fbd9fbbed98c615e14e8c2d/theme/colored/Nextcloud-macOS-icon.svg?short_path=44dd0e3)
> 
> This is a 100% vector Big-Sur-style icon, layer effects and all, and the above link should go to the code view, not the image preview.
> 
> (The annotation formatting apparently [broke the build process](https://github.com/nextcloud/desktop/pull/5757) in way that wasn't caught by the continuous integration, though...)
> 
> Anyway as you can see in the code annotation, I (as the author of the SVG document) released the template under a CC0 Public Domain Dedication, so hopefully this can be some help in creating a Big-Sur-style version of the RPI Imager icon!
> 
> (I should also note that the template is versatile enough that it should be useful for icon shapes other than the Apple "Squircle"—the annotations explain which SVG element to replace for a different shape.)

If you'd like to use this as template, it should be substantially closer to the Apple guidelines.

Ferbez, 2024-10-15:
I made an improved icon using Apple's [Design Resources](https://developer.apple.com/design/resources/) - #949.