raspberrypi/rpi-imager #747: download page (imager_latest.dmg) is stuck at 1.8.1 and 1.8.3 is only available via GitHub

core-code, 2023-11-30:
is the download page stuck at 1.8.1 on purpose?

https://www.raspberrypi.com/software/

lurch, 2023-12-02:
FYI
https://downloads.raspberrypi.org/os_list_imagingutility_v2.json
https://downloads.raspberrypi.org/os_list_imagingutility_v3.json
https://downloads.raspberrypi.org/os_list_imagingutility_v4.json

all still say `"latest_version": "1.7.4",` ?

AmnaSN, 2023-12-06:
The download page is still stuck at 1.8.1 for Mac

framps, 2023-12-10:
I just install latest bookworm desktop image and there ist rpi-image 1.8.3 is included.

core-code, 2023-12-10:
i am referring to this link

https://downloads.raspberrypi.org/imager/imager_latest.dmg

others may be fine

suschizu, 2023-12-11:
the website download still downloads version 1.8.1 instead of version 1.8.3 - is there a reason why the website link wasn't yet updated to version 1.8.3?

tdewey-rpi, 2023-12-11:
No good reason, just illness taking longer to resolve than hoped. This is on my list for this week, alongside other website updates.

suschizu, 2023-12-11:
ah ok sorry - get well soon :/

tdewey-rpi, 2023-12-13:
The website now points to 1.8.3, and I'm about to deploy a change to the manifests to prompt people to upgrade.

closed 2023-12-13 by tdewey-rpi