raspberrypi/rpi-imager #748: [BUG] Storage filtering incorrectly removes NVMe targets

tdewey-rpi, 2023-12-01:
**Describe the bug**
The 'Choose Storage' button presents a window that contains a curated list of storage targets to write OSes to. That curation incorrectly removes NVMe targets on CM4 and other NVMe-capable hardware.

**To Reproduce**
Using Imager v1.8.3:
- Connect NVMe SSD over PCIe
- Open Imager
- Select Raspberry Pi Device, OS, and then select 'Choose Storage'
- Observe no NVMe SSD present in the list

**Expected behaviour**
The NVMe SSD should be present in the list.

**Screenshots or video**
Not required.

**Desktop (please complete the following information):**
Raspberry Pi OS

**The name of the OS you are trying to write**
Not required

**Are you using OS Customisation?**
Not required

**Additional context**
Can be overriden by invoking `rpi-imager` with `--enable-writing-system-drives` as a command line argument

lurch, 2023-12-02:
Huh, is this a re-occurrence of #158 ?

HankB, 2023-12-02:
Yes, Looks like the same issue described in #158.

maxnet, 2023-12-02:
Unable to reproduce on Ubuntu:

[image]

HankB, 2023-12-03:
> Unable to reproduce on Ubuntu:

I can confirm that. I installed Ubuntu 23.10 (desktop) on an SD card when when I install and run `rpi-imager` the NVME SSD does appear as a storage target.

The version on Ubuntu 23.10 is `1.7.5+noembed-0ubuntu1 arm64` whereas the version on the 2023-11-28 RpiOS release is `1.8.3`.

tdewey-rpi, 2023-12-04:
Thanks for the update, @HankB 

This is fairly confusing, then. We've not materially changed the storage filtering mechanism since 1.7.5, so I wouldn't have expected a regression.

So I understand:

1. 1.8.3 is OK on Ubuntu
2. 1.8.3 is NOT OK on Raspberry Pi OS 2023-11-28
3. 1.7.5 is OK on Ubuntu

It sounds suspiciously like the bug is specific to Raspberry Pi OS? In which case, I'll try and find some hardware to reproduce your situation. In the meantime, could you tell me the device name of the NVMe storage in your set-up in Raspberry Pi OS?

closed 2023-12-05 by tdewey-rpi

samlearner-prog, 2024-06-28:
Hi - I was trying to use imager on Ubuntu 24.04 to create a raspios 64 image on an NVME drive using imager 1.8.5, and am getting the same issue - the NVME drive is not recognised.

HankB, 2024-06-28:
You need to differentiate between "not recognized by the OS" vs. "not listed by the imager." There are problems with some SSDs not working with RpiOS. Pimoroni has a list at <https://shop.pimoroni.com/products/nvme-base?variant=41219587178579>. A further note is that just because an SSD worked at one point does not mean it will still work.

Please list the output of `ls -l /dev/nvme*`

the-medo, 2024-08-14:
Hello, I'm having the same problem on Windows 11.

I'm new to Raspberry, but purchased Pi 5 with Samsung 980 SSD (listed in the link above that it should work). I would like to install ubuntu on it, so I downloaded latest Pi Imager for windows (v1.8.5)

To set it up, I connected SSD to my PC and set it up in bios. Windows can see that disk without a problem and it works, but my "Storage" window in imager is empty. I've tried a lot of stuff - conversions to Basic / Dynamic / GPT disks, with / without volumes, also went to bios and tried to hot-plug it, so now it is like a "removable drive".

I've just quickly checked the code, and PCIe/NVMe isn't listed under recognized bus types:
https://github.com/raspberrypi/rpi-imager/blob/7725ce7a4054371f9152cd0657f071bbbb9bb3ea/src/dependencies/drivelist/src/windows/list.cpp#L322C1-L342C2

Maybe that's why I can't see my SSD?