raspberrypi/rpi-imager #749: Customisation: Configure wireless LAN doesn't work on bookworm

thefool, 2023-12-07:
**Describe the bug**
wireless LAN does not get set up correctly on my raspberrypis, despite using Customisation
Culprit: `imager_custom: 127: uuid: not found`

**To Reproduce**
host OS: macOS Sonoma 14.0
Raspberry Pi Imager: v1.8.1
Raspberry Pi: Zero 2 W (not used as filter)
Raspberry Pi OS: Lite (64 bit) Bookworm, Released 2023-12-05

1. Choose the above OS in Imager
2. Customize OS 
  - Set hostname
  - Set username, password
  - Configure wireless LAN: MYSSID + MYPASSWORD, DE
  - Set locale settings
  - Enable SSH with password authentication
3. write to storage
4. boot RPi
5. doesn't connect to Wifi
6. connect a USB Ethernet Adapter:
  - [x] hostname was set
  - [x] username and password were set
  - [x] locale was set
  - [x] ssh works
  - [ ] wifi was NOT set 

**Expected behaviour**
It should connect to my Wifi, so I can connect to it via ssh

**Desktop (please complete the following information):**
host OS: macOS Sonoma 14.0

**The name of the OS you are trying to write**
Raspberry Pi OS: Lite (64 bit) Bookworm, Released 2023-12-05

**Are you using OS Customisation?**
Yes
  - Set hostname
  - Set username, password
  - Configure wireless LAN: MYSSID + MYPASSWORD, DE
  - Set locale settings
  - Enable SSH with password authentication

**Additional context**

I spent hours and hours trying to find a viable alternative to the WPA_SUPPLICANT method I have been using for years to set up a headless Raspberrypi zero W. This time, a new Zero 2 W. 

there's a lot of people complaining, online, but no-one has an answer, and I think I have found why:

## There's a bug in `/usr/lib/raspberrypi-sys-mods/imager_custom`!

1. Imager seems to correctly write `firstrun.sh` - I'll just paste the relevant section:
```
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan MYSSID' 'MYPASSWORDHASH' 'DE'
else
cat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'
country=DE
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
ap_scan=1

update_config=1
network={
	ssid="MYSSID"
	psk=MYPASSWORDHASH
}

WPAEOF
   chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf
   rfkill unblock wifi
   for filename in /var/lib/systemd/rfkill/*:wlan ; do
       echo 0 > $filename
   done
fi
```
2. `firstrun.sh` is executed and deleted as expected.
3. `/usr/lib/raspberrypi-sys-mods/imager_custom` does exist so it gets called. 
4. `imager_custom` fails: after having gained ssh access using ethernet, I tried running the command directly:
```
user@hostname:~ $ /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan -p MYSSID MYPASSWORD
/usr/lib/raspberrypi-sys-mods/imager_custom: 127: uuid: not found

user@hostname:~ $ sudo /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan -p MYSSID MYPASSWORD
/usr/lib/raspberrypi-sys-mods/imager_custom: 127: uuid: not found

user@hostname:~ $ sudo /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan MYSSID MYPASSWORDHASH DE
/usr/lib/raspberrypi-sys-mods/imager_custom: 127: uuid: not found
```

## SOLUTION:
instead of the above block of firstrun.sh, I wrote this into firstrun.sh:
```
#ADD WIFI CONFIG
#IF imager_custom present, use that, otherwise save config file.
#imager_custom seems to have an issue, so we'll comment that out

#if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
#   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'MYSSID' 'MYPASSWORDHASH' 'DE'
#else
cat >/etc/NetworkManager/system-connections/MYSSID.nmconnection <<'NMEOF'
[connection]
id=MYSSID
uuid=e3fa80e4-20e6-4411-8148-cc667b994996
type=wifi
interface-name=wlan0

[wifi]
mode=infrastructure
ssid=MYSSID

[wifi-security]
auth-alg=open
key-mgmt=wpa-psk
psk=MYPASSWORD

[ipv4]
method=auto

[ipv6]
addr-gen-mode=default
method=auto

[proxy]

NMEOF
   chmod 600 /etc/NetworkManager/system-connections/MYSSID.nmconnection
   rfkill unblock wifi
   for filename in /var/lib/systemd/rfkill/*:wlan ; do
       echo 0 > $filename
   done
#fi
```

This works well, but I'm guessing you should fix this:
`imager_custom: 127: uuid: not found`

thefool, 2023-12-07:
Just stumbled on this:
https://forums.raspberrypi.com/viewtopic.php?t=358046#p2165884
with an explanation what's causing the bug - a missing dependency!

lurch, 2023-12-08:
Thanks - created https://github.com/RPi-Distro/pi-gen/issues/735

maxnet, 2023-12-08:
>Thanks - created https://github.com/RPi-Distro/pi-gen/issues/735

Why in pi-gen and not just list uuid as dependency in the raspberrypi-sys-mods .deb where the calling script lives?
(There is already an issue in raspberrypi-sys-mods)

lurch, 2023-12-08:
ping @XECDesign for clarification, as he knows more about how all this works than I do :slightly_smiling_face:

XECDesign, 2023-12-11:
> ping @XECDesign for clarification, as he knows more about how all this works than I do 🙂

What Floris said is exactly right (https://github.com/RPi-Distro/raspberrypi-sys-mods/issues/86). Internal nightlies have had the fix since Friday. I didn't want to push out an image on a Friday, so there should be one later today.

lurch, 2023-12-12:
I believe the fix for this has been included in the new December 11th 2023 Lite images on https://www.raspberrypi.com/software/operating-systems/

beinbm, 2023-12-26:
I can confirm that its fixed in the new lite image from december

tdewey-rpi, 2024-01-05:
Closing as reported fixed.

closed 2024-01-05 by tdewey-rpi

SpottyMatt, 2024-01-31:
Confirming explicitly:

My Raspi Zero 2 W needed the December 11th, 23-bit, lite build ( `2023-12-11-raspios-bookworm-armhf-lite.img` / 9ce5e2c8c6c7637cd2227fdaaf0e34633e6ebedf05f1c88e00f833cbb644db4b ) in order to boot past the rainbow screen (64 bit builds stuck at rainbow screen) and past "finished configure lan for rpi-imager" on startup (32-bit builds Dec 5 & earlier hung there).

Notably, my Raspberry Pi Imager v1.8.4 was downloaded **before** the Dec 11th releases so it did not offer them - meaning I could try EVERY Raspbian recommended by the imager and none of them worked.

Thank you!

lurch, 2024-02-02:
> Notably, my Raspberry Pi Imager v1.8.4 was downloaded **before** the Dec 11th releases so it did not offer them

The list of OSes displayed by RPi Imager is downloaded dynamically (using https://downloads.raspberrypi.com/os_list_imagingutility_v4.json ), it isn't hardcoded inside Imager itself. So v1.8.4 will offer the same set of OSes as v1.8.5.

SpottyMatt, 2024-02-02:
That didn't seem to be the case; I launched the old imager many times over the course of a week and never got the updated ones.

However, today, The Dec11th releases are gone from my 1.8.5 imager. I guess the Dec 11 releases aren't in the JSON yet, but now I wonder how they ever showed up for me. Oh well, I guess that's neither here nor there.

lurch, 2024-02-03:
This is what I'm seeing in the JSON:
[image]
(Note that you'll need to select "No Filtering" if you want to write this image for a Zero 2 W)