raspberrypi/rpi-imager #754: Imager Password wrong flashing

Technorocker, 2023-12-16:
**Describe the bug**
For both versions I've used 1.8.1 and now I just installed 1.8.3 I kept getting access denied. After scraping the web for a bit I stumbled across some forum where the guy figured out the for whatever reason the first letter of your password you enter into the settings the first character doesn't get added in. So if my password is "raspberry" what the imager actually creates for the password is "aspberry"

**To Reproduce**
I JUST installed the newest version not 10 min ago and flash a pi zero w with buster. Same problem, I made my password, flashed the card, booted up the PI, connected using putty, tried password, access denied, tried password without the first character, access granted

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or video**
If applicable, add screenshots or a video to help explain your problem.

**Desktop (please complete the following information):**
windows 10

**The name of the OS you are trying to write**
BUSTER

**Are you using OS Customization?**
YES, this is where its bugged

nunomcardoso, 2023-12-16:
Can confirm. It just happened to me.

I was flashing two devices: Raspberry Pi 4 and Raspberry Pi Zero 2W
The Pi 4 worked flawlessly with Pi OS Lite 64bit (bookworm), and the password was exactly what I input.
The Pi Zero 2 W had me scratching my head for an hour before I found this issue. (Pi OS Lite Legacy 64Bit - bullseye)

`
> ssh user@zero2W
user@zero2W's password:
Permission denied, please try again.
`
The solution was not to type the first character like @Technorocker said.

After that, I reflashed (Pi Zero 2W) and typed everything in "OS Customisation", username and password again, WLAN password etc.
And plugged it to power. It surprised me that SSH worked with the full password this time, don't know what happened here.

Desktop: Windows 10
Imager version: 1.8.3

lurch, 2023-12-17:
Duplicate of #531 ?

Technorocker, 2023-12-17:
Actually yes, i just read over it and thats where i learned of this glitch a feq weeks ago  then the newest version came out and was still doing it i created this since i didnt recall where i had seen that originally.

tdewey-rpi, 2023-12-18:
To keep us on focus, closing this as a duplicate of #531.

closed 2023-12-18 by tdewey-rpi

lurch, 2023-12-18:
"To keep us on focus..." - nice pun :rofl: