raspberrypi/rpi-imager #755: [FEATURE] Supporting Larger Drives (GPT)

Technologyman00, 2023-12-16:
**Is your feature request related to a problem? Please describe.**
Building a gpt file system is a frustrating task for drives that are larger than 2TB

**Describe the solution you'd like**
I'd love to see rpi imager create a gpt partition tables to support larger 2TB partitions

**Describe alternatives you've considered**
One can build the partitions themselves but it's difficult or impossible without a Linux environment 

**Additional context**
It's possible to boot a pi with GPT partition table 
https://forums.raspberrypi.com/viewtopic.php?t=319435

maxnet, 2023-12-16:
May need more than just GPT to make large drives work.
E.g. chances are your drive has 4k sectors, which boot firmware.may not like.
https://github.com/raspberrypi/rpi-imager/issues/505#issuecomment-1342261358

Technologyman00, 2023-12-16:
My drives don't have 4k sectors. I'm just tired of doing it manually every time

lurch, 2023-12-17:
RPi Imager simply copies whatever's in your disk-image-file directly onto your SD-card / HardDrive. So if your disk-image contains MBR partitions, then you'll get MBR partitions on your HardDrvie. If you want your HardDrive to use GPT partitions, then you need to ensure that your disk-image contains GPT partitions. I.e. the decision whether to use MBR or GPT isn't anything to do with Raspberry Pi Imager.

[pi-gen](https://github.com/RPi-Distro/pi-gen) is the tool that's used to build the [Raspberry Pi OS images](https://www.raspberrypi.com/software/operating-systems/), and it creates MBR partitions for maximum compatibility. Perhaps you could [patch](https://github.com/RPi-Distro/pi-gen/blob/master/export-image/prerun.sh) `pi-gen` to create disk-images using GPT partitions instead?

> impossible without a Linux environment

If you don't have a Linux environment, I guess an easy workaround is to boot your Raspberry Pi from SD-card, use that to setup the GPT partitions on your HardDrive, then remove the SD-card and reboot.

maxnet, 2023-12-17:
>If you want your HardDrive to use GPT partitions, then you need to ensure that your disk-image contains GPT partitions

Things are not always that simple...

The GPT backup table live at end of drive...
So certainly it would need help from Imager if we want GPT support, as person creating image does not know how large user's disk is...

But I still think it is a moot point to implement, as long as boot firmware is not properly rewritten first to support 4k sectors first.
There may be large disks out there that still use 512 byte sectors as the TS mentioned. But the ones I have are 4k, and useless with Pi.

lurch, 2023-12-17:
> The GPT backup table lives at end of drive... (That's why Imager already also zeroes out end of drive before write).
> So certainly it would need help from Imager if we want standard compliant GPT support

Ah good point, I'd forgotten about that! :laughing:

guru, 2024-01-09:
Signal boosting this as I'm running a 4TB NVMe drive through the RPi 5 PCIe, and can only use 2TB as a boot drive using Imager. There doesn't seem to be any low-friction path for making anything larger than a 2TB a boot drive and using all the capacity. Sad Panda.

Technologyman00, 2024-01-10:
This is mostly because pi images come with a MBR partition table when you need GPT for higher than 2 TB. You also may run into block size issues for some larger drives. You can also try this, it's the easiest possible way https://forums.raspberrypi.com/viewtopic.php?f=29&t=196778

guru, 2024-01-10:
@Technologyman00 Thanks. Already went there and usb-boot rejected nvme* device as an invalid target. I got a solution for now, and it's *only* 20 simple steps ;)

As I say, signal boosting for a low-friction GPT (or glob forbid Hybrid) image available via Imager for a brighter future.

geerlingguy, 2024-01-15:
Agreed here — I've been getting emails from people buying NVMe HATs and trying to use 4 TB (or even 8 TB!) drives with the Pi 5, and running into the 2 TB size limit.

Outside of those intimately familiar with Linux, I would never recommend someone try messing with partitions, so there's effectively no simple way to use drives larger than 2TB on the Pi 5 right now for boot.

As mentioned above, there are a few tweaks that would be required for Imager, but whatever internal process is used at Pi Towers to generate the OS images would also need to be changed to support GPT (not sure where an issue for that would need to be created).

maxnet, 2024-01-15:
>As mentioned above, there are a few tweaks that would be required for Imager, but whatever internal process is used at 
>Pi Towers to generate the OS images would also need to be changed to support GPT (not sure where an issue for that 
>would need to be created).

Technically, Imager could also add the GPT automagically on large disks, and it could work if the OS is referring to its partitions by ext4 UUID.

Problem that it will only work with 512 byte drives and not 4K sector drives exists though.
That would need love from firmware folks to fix...

guru, 2024-01-15:
> Technically, Imager could also add the GPT automagically on large disks, and it could work if the OS is referring to its partitions by ext4 UUID.

This would be a big step forward. Getting a 4TB to work was a case of: partitioning via GPT, copying over the OS to each partition, updating the PARTUUIDs in the bootfs cmdline and rootfs fstab.

lurch, 2024-01-15:
Interestingly, https://www.google.com/search?q=nvme+4K+sector suggests that some SSDs can be switched between 512 byte sectors and 4KB sectors - I never knew that!

Technologyman00, 2024-01-15:
I sent a message through the contact us on the raspberry pi website with a link to this GitHub issue. No idea if that will go anywhere but hopefully it helps something.

ralphwiles, 2024-01-19:
> Signal boosting this as I'm running a 4TB NVMe drive through the RPi 5 PCIe, and can only use 2TB as a boot drive using Imager. There doesn't seem to be any low-friction path for making anything larger than a 2TB a boot drive and using all the capacity. Sad Panda.

So are you saying that the the other 2TB is lost in this case, can't be used?  Or just not available as a boot drive?  I just bought a Pi 5 8GB and I was planning to get a 4TB SSD using a Geekwork X1001 HAT for the SSD... I was going to partition the 4TB down into something like a 32GB partition for the RP system and then make a couple of other partitions that would be less than 2TB each.  If this wouldn't work then my backup plan would be to use a 32GB micro SD for the OS and use the 4TB drive for the rest of the things I was going to do.

sagesolar, 2024-06-30:
Is this something that is likely to be looked into being supported officially? Looking online support for NVMe drives larger than 2TB is possible. Some example approaches include:

- [docs.google.com/document/d/1OC3QgkhNz0pnramNMNcNXRwQcf6y0tmDd7no6YroMtg/edit](docs.google.com/document/d/1OC3QgkhNz0pnramNMNcNXRwQcf6y0tmDd7no6YroMtg/edit)
- [https://forums.raspberrypi.com/viewtopic.php?t=364768](https://forums.raspberrypi.com/viewtopic.php?t=364768)

It seems to me that there will be an increasing number of end users wanting to use their RPi's for storage solutions that support bootable, more than 2TB NVMe's. Surely adding support for creating a GPT disk image for such use cases is a good idea?