raspberrypi/rpi-imager #757: [BUG] Order of YES and NO buttons differs between popups, causing confusion.

SkyLined, 2023-12-20:
**Describe the bug**
* The [MsgPopup.qml](https://github.com/raspberrypi/rpi-imager/blob/qml/src/MsgPopup.qml) file has a [NO](https://github.com/raspberrypi/rpi-imager/blob/qml/src/MsgPopup.qml#L98) and a [YES](https://github.com/raspberrypi/rpi-imager/blob/qml/src/MsgPopup.qml#L107) button, in that order.
* The [UseSavedSettingsPopup.qml](https://github.com/raspberrypi/rpi-imager/blob/qml/src/UseSavedSettingsPopup.qml) file has a [YES](https://github.com/raspberrypi/rpi-imager/blob/qml/src/UseSavedSettingsPopup.qml#L125) and a [NO](https://github.com/raspberrypi/rpi-imager/blob/qml/src/UseSavedSettingsPopup.qml#L134) button, in that order.

The different order makes it hard to step through these popups if they are shown in succession; if you see YES and NO in one popup, you expect the same order in the next. I've repeatedly clicked NO in the second popup even though I wanted to click YES because I expected the buttons to be in the same order.

**To Reproduce**
Write Raspberry Pi OS to a device; the imager will ask "Would you like to apply OS customization settings?" using the `UseSavedSettingsPopup.qml` template (order = YES, NO). If you choose YES or NO, it will ask you "Are you sure you want to continue?" using the `MsgPopup.qml` template (order = NO, YES).

**Expected behaviour**
I expect consistency in the order of GUI components across the application.

**Screenshots or video**
[image]
[image]

**Desktop:**
 - OS: Windows 11 (but should be irrelevant)

**The name of the OS you are trying to write**
- Any Raspberry PI OS (but should be irrelevant)

**Are you using OS Customisation?**
No ( (but should be irrelevant)

**Additional context**
N/A

lurch, 2023-12-21:
See also #624

tdewey-rpi, 2023-12-21:
Thanks for the report, @SkyLined.

While reordering the buttons would be the most straightforward route, I suspect the better option is to actually redesign both dialogs.

I'm generally not a fan of yes/no dialogs. They require the user to have actually read the dialog proposition, which as is the case in this report, the user doesn't always do - instead intuiting actions based on trained behaviours from their historical system use.

I'll have a chat with our UI designers, but my preference would actually be to replace the row-of-buttons-with-question format with a stack-of-options format dialog. In the case of the second dialog, rather than [No | Yes], it would become...

[Erase all contents of the drive and continue,
Do not erase the drive, cancel the write].

And in the case of the OS customisation dialog...

[Apply OS customisations and write,
Do not apply OS customisations and write
Edit the OS customisations]

Dropping the 'No, clear...' option entirely, in favour of moving it's functionality into the OS customisation dialog itself.

Unfortunately, however, I doubt I'll get time to look at this before considering wider UI changes, and these have to be balanced along with my other projects.

lurch, 2023-12-23:
Also links to #700

SkyLined, 2024-01-07:
@tdewey-rpi 
> While reordering the buttons would be the most straightforward route, I suspect the better option is to actually redesign both dialogs.
Let's not let perfection get in the way of good. swapping the buttons should be a simple, straightforward change and would solve the immediate issue.

tdewey-rpi, 2024-02-07:
> @tdewey-rpi
> 
> > While reordering the buttons would be the most straightforward route, I suspect the better option is to actually redesign both dialogs.
> > Let's not let perfection get in the way of good. swapping the buttons should be a simple, straightforward change and would solve the immediate issue.

Apologies, I only realised you'd made a comment when re-reading this - looks like the comment formatter went a little wild.

If the argument was not one that would potentially present a user with a surprising OS configuration, one that deviates from the documentation of the OS vendor, then I may take a more charitable view of the proposal.

However, that is _not_ what is being proposed here. If you press 'no', you're getting the OS _as the OS provider intended_. If you press 'yes' and are _not actively aware of the customisations you have configured_ you will get an OS that is configured in a manner you do not expect - which is potentially _much_ more harmful than going through the OOBE of an OS.

I'm now going to close this issue, but I'll continue to mark this dialog for removal in a future UI redesign. If you wish to make a proposal for that UI redesign, please raise a Feature Request.

closed 2024-02-07 by tdewey-rpi