raspberrypi/rpi-imager #758: SSH Login not working in MacOS Raspberry Pi Lite image

rulkens, 2023-12-20:
**Describe the bug**
When writing the image for Raspberry Pi OS and setting the SSH user/password in the settings, login through ssh command line fails. It says the password is incorrect. In the old 1.7 version of the imager this was working.

**Expected behaviour**
I would expect ssh access to work with the username provided.

**Screenshots or video**
If applicable, add screenshots or a video to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS

**The name of the OS you are trying to write**
Raspberry Pi OS Lite 64bit

lurch, 2023-12-21:
Is this possibly a duplicate of #531 ?

rulkens, 2023-12-21:
It seems so, let's hope the fix described in #531 will work!

closed 2023-12-21 by rulkens