raspberrypi/rpi-imager #759: [FEATURE] disable password adjustment for root user

ProtekDev, 2023-12-21:
As i figured out after hours of testing, the ssh for the default user "root" is not working, bc of security reasons ?!
Thats why i wonder, if its possible to adjust this Imager Tool, that a user with the name root is not allowed for setting up a password or at least give the user an hint.

Thanks !

ProtekDev, 2023-12-21:
and still the bug that the config doesnt get created (No user, no password at all).
So i have to connect a monitor and keyboard to the raspberry, bc this Flashing seems to not work..

tdewey-rpi, 2023-12-22:
Thanks for the report, @ProtekDev. Unfortunately, this is a per-distro choice. In this case, you probably tried to flash a Debian-based image, where the default sshd configuration is:

[PermitRootLogin prohibit-password]

As such, even though you can set the password, you will not be allowed to use it by the default sshd configuration of the OS you've selected. Depending on the OS, however, you may be able to log in as root using an SSH key - a function that is already available in Imager today.

I'll keep this one in the backlog, but I suspect the solution involves a OS list schema change - one that I'm reticent to make without more reports of this issue.

lurch, 2023-12-23:
@tdewey-rpi Ignoring the SSH config issue for a minute, I think that @ProtekDev might have been suggesting that RPi Imager shouldn't allow you to create a username of `root` ? Presumably there'll be _lots_ of other unexpected side-effects if the user created by Raspberry Pi Imager is named `root`? Or is there an obvious use-case I'm missing where you **would** want to still allow that? :thinking:

tdewey-rpi, 2023-12-23:
> @tdewey-rpi Ignoring the SSH config issue for a minute, I think that @ProtekDev might have been suggesting that RPi Imager shouldn't allow you to create a username of `root` ? Presumably there'll be [...]

On Raspberry Pi OS and common LSB distros, yes. However, I'm reticent to claim that I have an exhaustive knowledge of how OS vendors choose to structure their OS, and in the absence of that knowledge I'm inclined to **not** disallow setting your username to _root_

maxnet, 2023-12-23:
>Or is there an obvious use-case I'm missing where you would want to still allow that? 🤔

In the past -before OS vendors started to disallow direct root access by default-, using the "root" account did was popular by some sysadmins because it allowed easy remote file system access with sftp/scp.
The argument being that there is a need to backup system files that are not world readible, and one cannot "sudo" from a file transfer program...

Naturually that is bad security practice.
(And so is using passwords for remote access. There are very few excuses not to use public key authentication.)

tdewey-rpi, 2024-04-30:
Closing as 'Won't Fix'. While it is poor security practice to use the 'root' user, as @maxnet has pointed out this was a feature of older operating system distributions.

Breaking these vintage distributions is not something I'm prepared to do - but I may consider adding a hint to a future version of Raspberry Pi Imager.

closed 2024-04-30 by tdewey-rpi