raspberrypi/rpi-imager #762: [BUG] Cannot SSH with default credentials

dnanak, 2023-12-23:
**Describe the bug**
I’m currently experiencing problems trying to SSH with the default username and password. It was working earlier, but now I can’t even ping to my Raspberry Pi Zero W to see what IP Address even comes up anymore.

**To Reproduce** 
So I own a Raspberry Pi Zero W. I downloaded the RP Imager to flash the OS Lite legacy (Raspbian) onto my 8gb SD. I then created an ssh and wpa_supplicant.conf to live inside the root, so it can be launched when the Raspberry Pi is booting up. 

I then pinged to see if it’s working as expected, which it was doing. I then proceed to ssh using pi and the default password (raspberry), which kept coming up with “access denied”. I tried finding every solution out there, none of which is working. 

I then made changes in the UI, by adding the password that I wanted to use but it seems as though every time I make the change, it will reappear with a long string of encrypted values which I can’t seem to work out. The same also appears for the SSID which I don’t seem to understand either. 

 - OS: macOS Ventura 

**The name of the OS you are trying to write**

OS Legacy Lite 32 Bit

**Are you using OS Customisation?**

Yes I am. 

**Additional context**
I can’t even run pings anymore, even after rewriting the SD card to take the wpa_supplicant.conf and the ssh.

lurch, 2023-12-23:
There's a few things here:
* Raspberry Pi OS no longer comes with a default username and password - you **have** to configure these using RPi Imager
* Raspberry Pi OS now uses NetworkManager instead of wpa_supplicant - so again you now need to configure these using Imager
* There was a small bug around password-entry ( #531 ) which has now been fixed in Imager 1.8.4

lurch, 2023-12-23:
Ahh, I've just spotted you said you're using Raspberry Pi OS Legacy (Bullseye) - the change to NetworkManager only happened in Bookworm, so I guess you can ignore that part of my reply!

dnanak, 2023-12-23:
@lurch Hey man, should I still use the ssh and wpa_supplicant as well?

lurch, 2023-12-23:
@dnanak I'm afraid that I've been so focused on Bookworm for so long, that I can't remember how things used to work in Bullseye, sorry! I guess either try it yourself to see if things work, or you could try asking on https://forums.raspberrypi.com/
(Many of the Raspberry Pi engineers are now on holiday until next year :christmas_tree: )

dnanak, 2023-12-27:
The update did get it working in the end, but I still had to create ssh and wpa_supplicant to enable this. Thanks for the help!

closed 2023-12-27 by dnanak