raspberrypi/rpi-imager #763: [BUG] Bookworm not offered for RPi older than RPi4 in images 1.8.1, 1.8.3, 1.8.4

eddylangley, 2023-12-23:
**Describe the bug**
Selecting anything other than RPi 4 or RPi 5 as the Raspberry Pi Device causes old Bullseye images to be offered not the newer Bookworm images. https://www.raspberrypi.com/software/operating-systems/ is adamant that Bookworm works on all RPi models.

**To Reproduce**
Install 1.8.1, 1.8.3 or 1.8.4 on Windows 11. Launch the imager, click CHOOSE DEVICE,  select Raspberry Pi 1 as the Raspberry Pi Device. Click CHOOSE OS.

**Expected behaviour**
Expected to see Bookworm images - was offered Bullseye images.

**Screenshots or video**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11

**Additional context**
Selecting RPi 4 or 5 seems to cause it to offer Bookworm.
Manually installing Bookworm (by pointing it at an image downloaded from https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2023-12-11/2023-12-11-raspios-bookworm-armhf-lite.img.xz seems to work fine
Looks like this issue appears from the point the imager became aware of platform differences; perhaps earlier RPi's weren't supported in Bookworm at the time?

lurch, 2023-12-23:
Duplicate of #730

bsimmo, 2024-01-01:
Bookworm is not even offered in the alternatives.

The Lite version of Bookworm works happily on any of the Pi3 series so far, there is no desktop things to worry about. (wayland), it has done since beta testing it.
This seems really odd choices and this is a question people are regularly asking, perhaps get someone to blog an update or provide some reason why in PiImager or on the PiImager download page.

Also, as a side, the desktop version should of course _never_ be offered on the 512MB 3A+ and Zero2 as _recommended_, but that is true for bullseye too.  Unfortunately, it is, and it works pants and can often just lock up if you ever try a browser on there. The lite isn't offered as a recommended, though it is perfect for it.

Please for the new year, have a good run through of the OS's on the current selling Pi's and see for yourself.

lurch, 2024-01-01:
> The Lite version of Bookworm works happily on any of the Pi3 series so far, there is no desktop things to worry about. (wayland), it has done since beta testing it.

We've found that sometimes doing a big `apt upgrade` on Bookworm Lite on a Raspberry Pi Zero 2 W can cause it to run out of memory and lock up the whole OS. And I **believe** that's one of the reasons why we're not recommending Bookworm Lite for lower-memory devices yet.

tdewey-rpi, 2024-01-05:
@lurch That is my understanding too. We have identified a workaround, but applying it consistently was still a work in progress.

We've now addressed this particular concern multiple times, and provided engineering explanations why it's not currently recommended. As before, if you are prepared to weather the risks associated you can use _No filtering_ to bypass the filtering - but when you select a device there is an implication that the image will work well - and I am not prepared to make that implication while I'm aware of fatal bugs.

As such, I'm going to close this issue - there's nothing to be done from the rpi-imager project PoV, and work on Raspberry Pi OS continues (as it typically does).

closed 2024-01-05 by tdewey-rpi

fab1an, 2024-03-14:
Have you got any further details on the out-of-memory situation? Where it occurs, during upgrade of which packages?

henryptung, 2024-04-06:
@tdewey-rpi Regarding the apt memory issue, is that being tracked anywhere? Just want to understand the prior art there and where progress is/isn't so far.

tdewey-rpi, 2024-04-10:
@fab1an @henryptung 

This is entirely the incorrect forum for these requests. This is the issue page for Raspberry Pi Imager, not for Raspberry Pi OS. I will now lock this issue to prevent further misuse of this issue tracker.

My understanding is that there is a workaround for the apt upgrade problem, however the recommendation is still being withheld until we are able to deploy Wayland by default - at which point I'd expect to see a number of pre-Raspberry Pi 4 platforms migrate to Bookworm. Currently, while you can manually install the `labwc` compositor, there are performance concerns on pre-Raspberry Pi 4 devices which we may attempt to address in the future, but which cause a material experience degradation on those platforms - hence no recommendation.