raspberrypi/rpi-imager #764: [BUG]Raspberry Pi Imager v1.8.4 OS customization Edit not showing window

AndyRH1701, 2023-12-24:
**Describe the bug**
1.8.4 does not show the Edit Settings window when the window is on the secondary monitor

**To Reproduce**
Click edit settings button with the window on the secondary monitor.

**Expected behaviour**
The setting window to open

**Screenshots or video**
N/A

**Desktop (please complete the following information):**
 - OS: Windows 11 22H2

**The name of the OS you are trying to write**
Has failed on all OSs I tested
**Are you using OS Customisation?**
Yes
**Additional context**
N/A

tdewey-rpi, 2024-01-05:
Thanks for the report, @AndyRH1701.

There is a known behavioural trait in Qt, where the pop up remembers it's location and tries to draw there. I use a Windows 11 system with 3 displays in total, and often see it appear on the internal laptop screen rather than either of the monitors.

Do you see the window if you use the Task View? (Windows Key + TAB)?

1337adde, 2024-06-07:
I also had this problem today on 1.8.5 with Windows 10. Unplugging my second monitor helped, the settings window was nowhere to be found.

ravirajchilka, 2024-06-09:
> I also had this problem today on 1.8.5 with Windows 10. The settings window never opened until I unplugged my second monitor.

I unplugged the monitor too, but still cant see anything when clicked on EDIT SETTINGS

prilr, 2024-09-12:
Same behavior on Win 10, v1.8.5, the settings window straight up does not appear at all if the Imager window is on the second monitor.

tdewey-rpi, 2024-09-12:
I'm not going to reopen this, as it's aimed at v1.8.5 - I've just released v1.9.0 on the 'Releases' page of this repository. If you're still seeing the problem with that release, please raise a new issue.