raspberrypi/rpi-imager #765: Imager

roberta-coloradogirl, 2023-12-24:
**Describe the bug**
Under Raspberry Pi OS on PI 4 and Pi 5after writing an OS image to ssd an error message appears after verify: "partition not found"

**To Reproduce**
write a custom image to a SSD

**Expected behaviour**
No errors

**Screenshots or video**
If applicable, add screenshots or a video to help explain your problem.

**Desktop (please complete the following information):**
Raspberry Pi  OS on Pi 4 or Pi 5 after sudo apt update && sudo apt full-upgrade

**The name of the OS you are trying to write**

**Are you using OS Customisation?**
No

**Additional context**
Add any other context about the problem here.

maxnet, 2023-12-24:
>write a custom image

What is the custom image you are using?

roberta-coloradogirl, 2023-12-25:
Armbian, slackware, free BSD. Might be that I tried to use customizations.