raspberrypi/rpi-imager #767: Error downloading: HTTP/2 stream 1 was not closed cleanly:[BUG]

rightsaidfred99, 2023-12-26:
**Describe the bug**
This bug has existed for years

**To Reproduce**
Just use raspberry pi imager

**Expected behaviour**
The program should download and install on the sd card, but it won't

**Screenshots or video**
[image]
**Desktop (please complete the following information):**
macOS14

**The name of the OS you are trying to write**
64bit

**Are you using OS Customisation?**
Yes / ssh and wifi prefill

**Additional context**
No issues with other pograms or downloads, just Pi Imager.

rightsaidfred99, 2023-12-26:
Whats worse, is it happens at 98% or 64% or 32%, but always enough to waste your time everytime you need to use this program

tdewey-rpi, 2023-12-26:
Thanks for the report, @rightsaidfred99 

Unfortunately you've not provided enough information for me to examine the problem. 

At first glance, it would appear the server hosting the OS you are trying to write has a misbehaving HTTP server. With that in mind, I would have contacted the OS vendor, however as multiple 64 Bit operating systems are offered in the OS selection, I cannot discern who to contact.

Could you please provide additional context - in particular:

- the precise OS you are trying to write
- if this issue occurs only with this OS
- If you have any kind of third party firewall, or atypical connection to the internet (eg, are you behind a corporate firewall, in university halls, etc)
- If you are using a hand-built router, one supplied by your ISP, or a commercially supported third party system (eg, a UniFi Dream Machine).

Thanks,

Tom

maxnet, 2023-12-26:
>At first glance, it would appear the server hosting the OS you are trying to write has a misbehaving HTTP server. With that 
>in mind, I would have contacted the OS vendor, however as multiple 64 Bit operating systems are offered in the OS 
>selection, I cannot discern who to contact.

Do note that server IP is mentioned in screenshot...

```
$ host downloads.raspberrypi.org
downloads.raspberrypi.org is an alias for lb.raspberrypi.org.
lb.raspberrypi.org is an alias for lb.raspberrypi.com.
lb.raspberrypi.com has address 93.93.135.117
lb.raspberrypi.com has address 46.235.231.145
lb.raspberrypi.com has address 46.235.231.111
lb.raspberrypi.com has address 176.126.240.84
lb.raspberrypi.com has address 46.235.227.39
lb.raspberrypi.com has address 93.93.135.118
lb.raspberrypi.com has address 46.235.231.151
lb.raspberrypi.com has address 93.93.130.212
lb.raspberrypi.com has address 176.126.240.86
lb.raspberrypi.com has address 46.235.230.122
lb.raspberrypi.com has address 93.93.135.141
lb.raspberrypi.com has address 176.126.240.167
lb.raspberrypi.com has IPv6 address 2a00:1098:84:1e0::1   <- Looks familiar
lb.raspberrypi.com has IPv6 address 2a00:1098:82:47::2:1
lb.raspberrypi.com has IPv6 address 2a00:1098:80:56::3:1
lb.raspberrypi.com has IPv6 address 2a00:1098:82:47::1
lb.raspberrypi.com has IPv6 address 2a00:1098:88:26::1:1
lb.raspberrypi.com has IPv6 address 2a00:1098:80:56::2:1
lb.raspberrypi.com has IPv6 address 2a00:1098:80:56::1:1
lb.raspberrypi.com has IPv6 address 2a00:1098:84:1e0::2
lb.raspberrypi.com has IPv6 address 2a00:1098:82:47::1:1
lb.raspberrypi.com has IPv6 address 2a00:1098:88:26::1
lb.raspberrypi.com has IPv6 address 2a00:1098:84:1e0::3
lb.raspberrypi.com has IPv6 address 2a00:1098:88:26::2:1
```

Looks like something closed the connection without saying bye over HTTP/2.
You may want to change the code to treat this as a non-fatal error that should trigger the retry mechanism instead of fatal.

maxnet, 2023-12-26:
Can reproduce by starting Imager from terminal, and simulating a connection stall by pausing program by pressing CTRL+Z in terminal, "sleep 61 ; fg"

[image]

If the download stalls for over a minute the server chucks you out by closing TCP connection, and that gives that message if running OSX.

Here's a build that does retry in such cases: [Raspberry Pi Imager 1.8.5.dmg](https://www.maxnet.eu/tmp/1.8.5/Raspberry%20Pi%20Imager%201.8.5.dmg)

Do keep in mind that if your connection stalls that long, that does is a sign you may have some other deeper problems, like bad Internet connection or local wifi...

danesparza, 2024-01-05:
@maxnet I can confirm that the 1.8.5 build you linked to works great.  1.8.4 and 1.8.3 were causing issues on my MacOS machine.  

I have AT&T fiber internet, a gigabit switch, Wifi-6 access points, and Unifi running everything.  I routinely download large files.  My internet connection is not the issue.  

Please include this fix in future releases.

tdewey-rpi, 2024-01-05:
Fix will be included in 1.8.5.

closed 2024-01-05 by tdewey-rpi

softlybutch, 2024-07-18:
> Can reproduce by starting Imager from terminal, and simulating a connection stall by pausing program by pressing CTRL+Z in terminal, "sleep 61 ; fg"
> 
> ![Screenshot 2023-12-26 at 15 59 34](https://private-user-images.githubusercontent.com/1123701/292887944-865953ed-58d7-40f9-bbe7-1f53a0124e72.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzMzI0NDEsIm5iZiI6MTcyMTMzMjE0MSwicGF0aCI6Ii8xMTIzNzAxLzI5Mjg4Nzk0NC04NjU5NTNlZC01OGQ3LTQwZjktYmJlNy0xZjUzYTAxMjRlNzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MThUMTk0OTAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjdmYWJhY2VmZDE0Y2UwZjRmN2M5MWNjMTY1ZjIzZWM4ZGMwZWE0NDVmNTYwNDBjZTU5MTVlZGFlZDljYjA3NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yNoaoXvrgfN8wDWJmcw6oQyz1P0SlY_0BXdGtjfhubE)
> 
> If the download stalls for over a minute the server chucks you out by closing TCP connection, and that gives that message if running OSX.
> 
> Here's a build that does retry in such cases: [Raspberry Pi Imager 1.8.5.dmg](https://www.maxnet.eu/tmp/1.8.5/Raspberry%20Pi%20Imager%201.8.5.dmg)
> 
> Do keep in mind that if your connection stalls that long, that does is a sign you may have some other deeper problems, like bad Internet connection or local wifi...

I'm getting this error as well in ubuntu, is there a similar solution?