raspberrypi/rpi-imager #769: [BUG] Windows 10, Storage capacity is not large enough 127.8 GB sd card vs 127.9 GB image

BdR76, 2023-12-27:
**Describe the bug**
I've bought a new 128GB sd card and got hold of a pre-build 128GB image, but when I try to write the image to the sd card, Raspberry Pi Imager gives the error message "Storage capacity is not large enough. Needs to be at least 127.9 GB".

It's a brand new Micro SD card (SanDisk microSDXC UHS-I 140MB/s), and I tried two different pre-build 128GB images, both with the exact same result.

Now I know SD cards can have slightly less memory capacity than what it says on the packaging, but is this really a SD card capacity issue?

Could it be caused by Windows reporting the sd memory capacity differently? According to [this page](https://bulkmemorycards.com/why-is-my-flash-drive-not-showing-full-storage-capacity/):

> The reported storage capacity from your operating system is different from the storage capacity on the packaging or label because of the difference between the binary and decimal systems of measurement. The product manufacturer measures in base 10. The base 10 measurements are named MegaByte, GigaByte, TeraByte, etc. Windows 10/11 measures in base 2. The base 2 measurements are named MebiByte, GibiByte, TebiByte, etc.
> 
> In the example pictured below of the same 128GB SD card, Windows 10 on the left and MacOS on the right, you see the difference. The label on the card is 128GB (base 10); Windows 10 displays a 119GB capacity (base 2). **However, you will notice it is saying there are 127,831,900,160 bytes.** On a base 10 system, this would read as **127.8GB**, which is way closer to the advertised 128GB. This means that the labeled size of the drive **_IS_** the size of the drive. Your Windows 10/11 computer is just measuring it differently. **_There is no size discrepancy._**
> ..
> But 127.8GB on either OS is still not the 128GB you paid for! So, what else is going on?

So if I understand correctly it might have to do with how Windows 10 interprets the SD capacity? Is this value that Windows reports the value that Raspberry Pi Imager checks against internally? Could that be causing the issue?

**To Reproduce**
Include steps to allow us to reproduce the error. Include your host OS, and the version of Raspberry Pi Imager.

1. Get a pre-build 128GB Raspberry Pi4 batocera image
2. Get a brand new 128GB SD card
3. Start Raspberry Pi Imager v1.8.4
4. Choose device: Raspberry Pi4
5. Choose OS: custom image
6. Choose storage: SD card in SD slot
7. Apply OS customisation settings: No8. 
9. Error message "Storage capacity is not large enough"

**Expected behaviour**
I expected a 128GB image to fit on a 128GB SD card

**Screenshots or video**
Here are some screenshots

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Home 19045.3803
 - Raspberry Pi Imager v1.8.4

**The name of the OS you are trying to write**
Batocera V36

**Are you using OS Customisation?**
no

**Additional context**
No additional context

BdR76, 2023-12-27:
Btw I've seen other users [run into the exact same issue](https://www.reddit.com/r/cade/comments/10ilv3i/microsd_card_help_1278_vs_1279/).

And also, if this _is_ a memory capacity issue, then does that mean you have to buy a 256GB card to ensure you can write a 128GB image, a 512GB card for a 256GB image and so on? That seems quite odd.

lurch, 2023-12-27:
> Now I know SD cards can have slightly less memory capacity than what it says on the packaging, but is this really a SD card capacity issue?

I'm afraid that this is _exactly_ your problem. Unfortunately an SD card bought from one manufacturer labelled as "128GB" might have more or less actual memory capacity than a "128GB" card purchased from another manufacturer.
So in your particular case, it sounds like the pre-built image you're downloading was created on a 127.9GB card, and the card that you've purchased is "only" 127.8GB.

> And also, if this is a memory capacity issue, then does that mean you have to buy a 256GB card to ensure you can write a 128GB image, a 512GB card for a 256GB image and so on?

Yes. Alternatively, the people producing your pre-built image could shrink it down first (e.g. using https://github.com/Drewsif/PiShrink ) before uploading it, so that it'd then be small enough to be written to **any** "128GB" card.

BdR76, 2023-12-27:
Thanks for the reply, btw I tried writing the same 128GB image using Win32 Disk Imager, it has a checkbox option `Read Only Allocated Partitions` which I assume skips empty sectors. Unfortunately it runs into the same capacity problem.

> More space required than is available:
> Required: 249737216 sectors
> Available: 249670656 sectors
> Sector size: 512
> 
> The extra space DOES appear to contain data
> 
> Continue Anyway?

So yes I think you're right, the ideal solution would be that the people creating these pre-built images pre-shrink the image before uploading it, to accommodate the varying SD card capacities.

BdR76, 2023-12-29:
fyi I also tried writing the 128GB image with balenaEtcher with the same result

> Too small! The selected source is 34.1 MB larger than this drive

It really does seem to be the case that the image simply is _slightly_ larger than a new 128GB sd card, so I guess this issue can be closed.

lurch, 2023-12-30:
> the image simply is _slightly_ larger than a new 128GB sd card

The image is slightly larger than _some_ new 128GB SD cards; not necessarily larger than **all** new 128GB SD cards :wink:

BdR76, 2024-01-19:
Still though, seems a little weird that you have to find the exact same brand SD card to make sure you can write a backed-up image. Even if you could find the exact same micro SD brand (there are quite a lot) isn't it possible that the capacity could still be slightly off by a small amount (idk just speculating) ?

Like you say, those who create the pre-built image could shrink it down first, I've created an illustration, that's about accurate to this situation right?

[image]

lurch, 2024-01-19:
> I've created an illustration, that's about accurate to this situation right?

Yeah that's a good illustration of what's going on. If you were lucky, then the backed-up-image could have been created from card C, and then RPi Imager would have been able to write it to all other "128GB" cards; but if you were unlucky the backed-up-image would have been created from card X, meaning that it won't fit on most other cards.