raspberrypi/rpi-imager #770: [BUG] Insecure permissions for configuration file holding hashed password & plaintext WiFi password

ricardobranco777, 2023-12-30:
**Describe the bug**

The configuration file which holds the hashed & plaintext WiFi has insecure permissions by default.

**To Reproduce**

on Linux:

```
$ ls -l "$HOME/.config/Raspberry Pi/Imager.conf"
-rw-rw-r--. 1 ricardo ricardo 775 Nov 19 12:13 Imager.conf
$ grep Password "$HOME/.config/Raspberry Pi/Imager.conf"
sshUserPassword=$5$xxx
wifiPassword=xxx
```

**Expected behaviour**

0600 permissions.

audas, 2024-03-02:
Was under the impression that if we just re-used the passwords that were set in the imager that they would be correct - but they are not working for SSH.

So if I re-use a password for another Pi it is wrong as it has been hashed.

However the WiFi password remains correct???

tdewey-rpi, 2024-03-04:
> Was under the impression that if we just re-used the passwords that were set in the imager that they would be correct - but they are not working for SSH.
> 
> So if I re-use a password for another Pi it is wrong as it has been hashed.
> 
> However the WiFi password remains correct???

@audas This sounds like a different problem, but also one that I would expect to be true, as I don't know if SSH uses the same hashing scheme as WPA-PSK. Please raise unique issues for unique problems.

tdewey-rpi, 2026-10-21:
2.0.12 shipped with the fix for this.

closed 2026-10-21 by tdewey-rpi