raspberrypi/rpi-imager #774: [BUG] Application fails at startup: qrc:/main.qml:681:5: Cannot assign object to list property "contentData"

yurivict, 2024-01-05:
**Describe the bug**
```
QQmlApplicationEngine failed to load component
qrc:/main.qml:681:5: Cannot assign object to list property "contentData"
```

**To Reproduce**
Run rpi-imager.

**Expected behaviour**
n/a

**Screenshots or video**
n/a

**Desktop (please complete the following information):**
 - OS: FreeBSD 14.0

**The name of the OS you are trying to write**
n/a

**Are you using OS Customisation?**
Patched out WiFi-related parts.

maxnet, 2024-01-06:
Seems to be a problem that only affects the Qt 5.15 FreeBSD package.
Get the same error as you if I use that, but if I do "pkg install qt6" and turn the CMakeLists.txt around to prefer Qt 6 over 5 if both are found, then it does work work fine.

[image]

At least to the point of starting.
Will leave the other problems like fixing disk enumeration (on Linux we outsource that to lsblk, but the lsblk package FreeBSD has does not have the "--json" option we use), and finding solutions to open raw disk devices as non-root user under FreeBSD, and unmounting storage if it is already mounted to you.

maxnet, 2024-01-06:
Hmm, while it does start under Qt 6, the operating system window shows an empty list (even though if I add a console.log(osmodel.count) to the onClick of the "choose os" button it does print that it has 12 items).
So something with that wrong as well.

Does run fine under Qt 6 on other operating systems. (Our OSX build is Qt 6, although we still default to 5 for other OS).
So may be something FreeBSD or Qt subrelease specific as well.

tdewey-rpi, 2024-09-11:
Unfortunately I don't have the bandwidth to support FreeBSD alongside the other major OSs we provide releases for.

We now use Qt6 on all platforms, and a later version of Qt6 at that - you may have success starting there for a FreeBSD port.

I'm prepared to accept PRs adding better FreeBSD support, but these would not see active maintenance from myself, and may be pruned later if they are not actively maintained.

'Won't fix'.

closed 2024-09-11 by tdewey-rpi