raspberrypi/rpi-imager #775: [FEATURE] add option to enable `usb_max_current_enable=1` for pi 5

godalming123, 2024-01-07:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I have to shut down windows, boot into Linux, mount the USB, and edit the max_current_enable setting just to use USB boot with my 5v 3a power supply.

**Describe the solution you'd like**
In the OS config menu, their should be an option to enable USB boot for 15 watt power supplies on the raspberry pi 5.

maxnet, 2024-01-07:
>I'm always frustrated when I have to shut down windows, boot into Linux, mount the USB, and edit the max_current_enable setting just to use USB boot with my 5v 3a power supply.

Eh, shouldn't Windows be able to access the FAT partition with config.txt as well?

godalming123, 2024-01-07:
Maybe that's possible, enabling USB boot would still be mildly annoying though.

tdewey-rpi, 2024-01-08:
I think @maxnet's point is that after writing, you could hotplug the USB storage and adjust `config.txt` from inside Windows - no Linux PC use required?

tdewey-rpi, 2024-02-05:
No further update in a month - closing as stale.

We may explore this for 2.0, however I can't immediately see how to expose this to new users without causing some degree of confusion.

closed 2024-02-05 by tdewey-rpi