raspberrypi/rpi-imager #776: [BUG] Error downloading: schannel: next InitializeSecurityContext failed: Unknown error (0x80092013)

SkyLined, 2024-01-07:
**Describe the bug**
While attempting to write any Raspberry Pi OS to storage, I get the following error popup:

[image]

Windows Error [0x80092013](http://ec.skylined.nl/#0x80092013) is [CRYPT_E_REVOCATION_OFFLINE](https://learn.microsoft.com/en-us/windows/win32/api/wincrypt/ns-wincrypt-cert_chain_policy_status).

I do not know how revocation servers are implemented, so I cannot provide detailed info. What I do know is that the IP address 176.126.240.86 responds to ping and has a [HTTP](http://176.126.240.86) server that redirects to the secure https://www.raspberrypi.org. It also has a [HTTPS](https://176.126.240.86) server that provides a certificate is for `*.raspberrypi.org`. When sending a request to the HTTPS server with a `Host: www.raspberrypi.org` header, I get a 200 OK response. In another attempt, I got the IP address 46.235.227.39, which has the exact same properties.

Given that I have had this issue for a few weeks now, I suspect there's something about my set-up that is causing it; otherwise more people would have reported it.

**To Reproduce**
* Select Raspberry Pi 4 as Device (I believe the exact device is irrelevant)
* Select Raspberry Pi OS (64-bit) as the OS (I believe the exact OS is irrelevant)
* Select a USB stick as storage (I believe the exact storage is irrelevant)
* Click next
You should see the error after a few seconds when it tries to download the OS. 

My system:
* 
* RPI Imager v1.8.3 and v1.8.4

**Expected behaviour**
Download and write the OS.

**Screenshots or video**
See above.

**Desktop (please complete the following information):**
 - Windows 11 Pro 21H2 (Version 10.0.22000.2538)

**The name of the OS you are trying to write**
The problem appears to affect all Raspberry Pi OS & OS Lite versions (32- and 64-bit) for all devices, but I have not exhaustively checked all of them.

**Are you using OS Customisation?**
Yes and No: tried both with the same result.

**Additional context**
N/A

SkyLined, 2024-01-07:
Note that I can manually download an image and ask imager to burn it as a work-around.

SkyLined, 2024-01-07:
[Others have experienced this issue as well.](https://forums.raspberrypi.com/viewtopic.php?t=362797)

That user chose to use other software to burn the image, suggesting that this issue may be under-reported.

maxnet, 2024-01-07:
>What I do know is that the IP address 176.126.240.86 responds to ping

That is the download server that do is the final target.
It is however complaining that it cannot contact the revocation server that is used to double check that our SSL certificate is in good standing and has not been revoked.

Can you reach http://r3.o.lencr.org/ ?

SkyLined, 2024-01-08:
No, I could not reach it: my pi-hole DNS server resolved it to 0.0.0.0. I found out I manually add it to the blocklist. I assume I did this after I noticed devices connecting to it for no obvious reason (I tned to lock things down hard).

After removing the entry, the issue is resolved. Thank you for the help!

If time allows, I would recommend improving the error message to provide additional details such as the domain name for the revocation server that cannot be contacted, as that might help others debug this.

closed 2024-01-08 by SkyLined

maxnet, 2024-01-09:
>If time allows, I would recommend improving the error message to provide additional details such as the domain name for the revocation server that cannot be contacted, as that might help others debug this.

I am afraid that is not trivial for us to do.
Imager outsources connectivity to libcurl who in turns outsources TLS to system libraries like schannel, OpenSSL or GnuTLS depending on operating system.

Due to all the abstractions involved all Imager gets to hear on error, is a generic error code indicating SOMETHING tls related failed like CURLE_SSL_CONNECT_ERROR, and the textual error message coming from the system library.
The textual error message not only differs by system library, but can also be different depending on the language of the system.
So it us hard for Imager to determine exact error and improve the message coming from the library...

Beauxrel, 2024-04-30:
I had this issue and solved it by disconnecting my VPN.

drph4nt0m, 2024-09-02:
> I had this issue and solved it by disconnecting my VPN.

+1. Had the same error and resolved by disconnecting my VPN.

kambagiri, 2024-11-10:
Thanks for sharing the post, disconnecting VPN and restarting the process worked for me in Raspberry Pi 5 and Raspberry Pi 2 devices !!!

uditp-resideo, 2024-11-18:
Disconnecting VPN works. Thanks