raspberrypi/rpi-imager #778: [BUG] Imager crashes on launch on Big Sur

devtimi, 2024-01-09:
**Describe the bug**
Imager 1.8.x crashes on launch on Big Sur.

**To Reproduce**
1. Have macOS 11
2. Launch imager.app

**Expected behaviour**
The app to launch and operate.

**Desktop (please complete the following information):**
 - OS: macOS 11.7.10 (20G1427)
 - ARM-64 (Native)

**Additional context**
Archive build 1.7.0 works as expected but tells me there's an update.

[rpi-imager.crash.txt](https://github.com/raspberrypi/rpi-imager/files/13877984/rpi-imager.crash.txt)

tdewey-rpi, 2024-01-10:
So, it appears to be exploding in the ctor for ImageWriter:

`1   org.raspberrypi.imagingutility	0x0000000102e6867c ImageWriter::ImageWriter(QObject*) + 364`

And with a SEGV:

`Exception Type:        EXC_BAD_ACCESS (SIGSEGV)`

Curious that we're not seeing this on other macOS versions. I'm going to take a look at the code deltas between HEAD and 1.7.0 to see if we can get a hint.

maxnet, 2024-01-10:
Not sure what the problem is.
If it crashes around +364 that seems to be after DriveListModel::DriveListModel() was called when constructing the ImagerWriter object.

```
lldb Contents/MacOS/rpi-imager
(lldb) target create "Contents/MacOS/rpi-imager"
Current executable set to '/Applications/Raspberry Pi Imager.app/Contents/MacOS/rpi-imager' (arm64).
(lldb) disas -n 'ImageWriter::ImageWriter'
rpi-imager`ImageWriter::ImageWriter:
rpi-imager[0x100094510] <+0>:     stp    x28, x27, [sp, #-0x60]!
rpi-imager[0x100094514] <+4>:     stp    x26, x25, [sp, #0x10]
rpi-imager[0x100094518] <+8>:     stp    x24, x23, [sp, #0x20]
rpi-imager[0x10009451c] <+12>:    stp    x22, x21, [sp, #0x30]
rpi-imager[0x100094520] <+16>:    stp    x20, x19, [sp, #0x40]
rpi-imager[0x100094524] <+20>:    stp    x29, x30, [sp, #0x50]
rpi-imager[0x100094528] <+24>:    add    x29, sp, #0x50
rpi-imager[0x10009452c] <+28>:    sub    sp, sp, #0x2f0
rpi-imager[0x100094530] <+32>:    mov    x25, x0
rpi-imager[0x100094534] <+36>:    adrp   x8, 340
rpi-imager[0x100094538] <+40>:    ldr    x8, [x8, #0xc8]
rpi-imager[0x10009453c] <+44>:    ldr    x8, [x8]
rpi-imager[0x100094540] <+48>:    stur   x8, [x29, #-0x60]
rpi-imager[0x100094544] <+52>:    bl     0x1000c36f4               ; symbol stub for: QObject::QObject(QObject*)
rpi-imager[0x100094548] <+56>:    adrp   x8, 341
rpi-imager[0x10009454c] <+60>:    add    x8, x8, #0xfc0            ; vtable for ImageWriter
rpi-imager[0x100094550] <+64>:    add    x8, x8, #0x10
rpi-imager[0x100094554] <+68>:    str    x8, [x0], #0x10
rpi-imager[0x100094558] <+72>:    str    x0, [sp, #0x50]
rpi-imager[0x10009455c] <+76>:    mov    x1, x25
rpi-imager[0x100094560] <+80>:    bl     0x1000c3400               ; symbol stub for: QNetworkAccessManager::QNetworkAccessManager(QObject*)
rpi-imager[0x100094564] <+84>:    add    x0, x25, #0x20
rpi-imager[0x100094568] <+88>:    str    x0, [sp, #0x38]
rpi-imager[0x10009456c] <+92>:    bl     0x1000c30ac               ; symbol stub for: QJsonDocument::QJsonDocument()
rpi-imager[0x100094570] <+96>:    add    x0, x25, #0x28
rpi-imager[0x100094574] <+100>:   str    x0, [sp, #0x30]
rpi-imager[0x100094578] <+104>:   bl     0x1000c2e30               ; symbol stub for: QJsonArray::QJsonArray()
rpi-imager[0x10009457c] <+108>:   add    x0, x25, #0x38
rpi-imager[0x100094580] <+112>:   str    x0, [sp, #0x28]
rpi-imager[0x100094584] <+116>:   bl     0x1000c3514               ; symbol stub for: QUrl::QUrl()
rpi-imager[0x100094588] <+120>:   adrp   x1, 332
rpi-imager[0x10009458c] <+124>:   add    x1, x1, #0xe29            ; "https://downloads.raspberrypi.org/os_list_imagingutility_v4.json"
rpi-imager[0x100094590] <+128>:   add    x8, sp, #0x178
rpi-imager[0x100094594] <+132>:   mov    w0, #0x40
rpi-imager[0x100094598] <+136>:   str    x25, [sp, #0x60]
rpi-imager[0x10009459c] <+140>:   bl     0x1000c37cc               ; symbol stub for: QString::fromUtf8(QByteArrayView)
rpi-imager[0x1000945a0] <+144>:   add    x0, x25, #0x40
rpi-imager[0x1000945a4] <+148>:   add    x8, sp, #0x79
rpi-imager[0x1000945a8] <+152>:   ldur   q0, [x8, #0xff]
rpi-imager[0x1000945ac] <+156>:   add    x8, sp, #0x59
rpi-imager[0x1000945b0] <+160>:   stur   q0, [x8, #0xff]
rpi-imager[0x1000945b4] <+164>:   ldr    x8, [sp, #0x188]
rpi-imager[0x1000945b8] <+168>:   str    x8, [sp, #0x168]
rpi-imager[0x1000945bc] <+172>:   add    x1, sp, #0x158
rpi-imager[0x1000945c0] <+176>:   str    x0, [sp, #0x80]
rpi-imager[0x1000945c4] <+180>:   mov    w2, #0x0
rpi-imager[0x1000945c8] <+184>:   bl     0x1000c34fc               ; symbol stub for: QUrl::QUrl(QString const&, QUrl::ParsingMode)
rpi-imager[0x1000945cc] <+188>:   ldr    x8, [sp, #0x158]
rpi-imager[0x1000945d0] <+192>:   cbz    x8, 0x1000945f4           ; <+228>
rpi-imager[0x1000945d4] <+196>:   mov    w9, #-0x1
rpi-imager[0x1000945d8] <+200>:   ldaddal w9, w8, [x8]
rpi-imager[0x1000945dc] <+204>:   cmp    w8, #0x1
rpi-imager[0x1000945e0] <+208>:   b.ne   0x1000945f4               ; <+228>
rpi-imager[0x1000945e4] <+212>:   ldr    x0, [sp, #0x158]
rpi-imager[0x1000945e8] <+216>:   mov    w1, #0x2
rpi-imager[0x1000945ec] <+220>:   mov    w2, #0x8
rpi-imager[0x1000945f0] <+224>:   bl     0x1000c2d28               ; symbol stub for: QArrayData::deallocate(QArrayData*, long long, long long)
rpi-imager[0x1000945f4] <+228>:   movi.2d v0, #0000000000000000
rpi-imager[0x1000945f8] <+232>:   mov    x8, x25
rpi-imager[0x1000945fc] <+236>:   str    q0, [x8, #0x48]!
rpi-imager[0x100094600] <+240>:   mov    x9, x8
[... 19 lines cut ...]
rpi-imager[0x100094650] <+320>:   stp    q0, q0, [x8, #0xe0]
rpi-imager[0x100094654] <+324>:   stp    q0, q0, [x8, #0xc0]
rpi-imager[0x100094658] <+328>:   stp    q0, q0, [x8, #0xa0]
rpi-imager[0x10009465c] <+332>:   stp    q0, q0, [x8, #0x70]
rpi-imager[0x100094660] <+336>:   stp    q0, q0, [x8, #0x40]
rpi-imager[0x100094664] <+340>:   stp    q0, q0, [x8, #0x10]
rpi-imager[0x100094668] <+344>:   stp    xzr, xzr, [x8, #0x180]
rpi-imager[0x10009466c] <+348>:   str    x8, [sp, #0x20]
rpi-imager[0x100094670] <+352>:   str    x0, [sp, #0x10]
rpi-imager[0x100094674] <+356>:   mov    x1, #0x0
rpi-imager[0x100094678] <+360>:   bl     0x1000a28e8               ; DriveListModel::DriveListModel(QObject*)
rpi-imager[0x10009467c] <+364>:   str    xzr, [x25, #0x210]
rpi-imager[0x100094680] <+368>:   add    x0, x25, #0x218
rpi-imager[0x100094684] <+372>:   str    x0, [sp, #0x58]
rpi-imager[0x100094688] <+376>:   mov    x1, #0x0
[... cut ...] Object*)
rpi-imager[0x100094690] <+384>:   add    x0, x25, #0x228
rpi-imager[0x100094694] <+388>:   str    x0, [sp, #0x68]
rpi-imager[0x100094698] <+392>:   mov    x1, #0x0
[....]
```

However DriveListModel() does not do much, and the code is not new.
Only thing that did change recently is the addition of the _networkManager stuff.
But that may not be the actual problem either. Could be that memory is corrupted somewhere else, but that it didn't show until now because it was not fatal before...

maxnet, 2024-01-14:
Problem with reproducing it is that it does not happen for me on other OSX versions, and I do not have anything that can run the ARM version of Big Sur.
My M2 Mac Mini is too new to downgrade to it, and refuses to install it from USB installation media: 

[image]

And UTM (virtualization platform) does not like it either:

[image]

Think running ARM OSX as VM guest is only supported starting with Montermery.

@devtimi
Anyway, here is a build with some good old fashioned debug print statements added (and LTO disabled):

[Raspberry Pi Imager 1.8.5.dmg](https://www.maxnet.eu/tmp/osx-problem/Raspberry%20Pi%20Imager%201.8.5.dmg)

Can you perhaps open the .DMG, drag Imager to your Applications folder, open a "Terminal" and execute:

```
/Applications/Raspberry\ Pi\ Imager.app/Contents/MacOS/rpi-imager
```

I would be interested in the last line that gets printed to console before it crashes.

devtimi, 2024-01-17:
Your build launched. Here's the output just in case that helps.

```
Before imageWriter
DriveListModel::DriveListModel start
DriveListModel::DriveListModel after rolenames
DriveListModel::DriveListModel end
Imagewriter::ImageWriter() start
After imageWriter
OSX most preferred language: "en_US"
qt.qpa.fonts: Populating font family aliases took 384 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
qt.tlsbackend.ossl: Failed to load libssl/libcrypto.
```

maxnet, 2024-01-19:
>Your build launched.

Does this one also still works?

[Raspberry Pi Imager 1.8.5.dmg](https://www.maxnet.eu/tmp/185-nolto/Raspberry%20Pi%20Imager%201.8.5.dmg)

(Removes debug statements again. But keeps Link Time Optimization disabled).

devtimi, 2024-01-21:
This version 1.8.5 also launches!

maxnet, 2024-01-21:
Good to hear. Then I'll close this.

closed 2024-01-21 by maxnet