raspberrypi/rpi-imager #779: [BUG] the 'run ssh-keygen' button in OS Customization should not be destructive

vovinhd, 2024-01-10:
**Describe the bug**
In the OS Customization dialog, in the services tab, checking 'Enable SSH' and selecting 'Allow public-key authentication only' did not prefill the set authorized_keys for ${USERNAME} with the content of my existing keypair stored as ./ssh/id_rsa.pub and ./ssh/id_rsa. Selecting 'run ssh-keygen' overwrote the existing keys at that location without prompting that this is what it might do, and that information would be lost. 

**To Reproduce**
- have an existing ./ssh/id_rsa.pub and ./ssh/id_rsa keypair 
- run the OS customization
- navigate to 'services' 
- check 'Enable SSH'
- select  'Allow public-key authentication only'
- notice the empty field
- select 'run ssh-keygen' (because using different keys for different hosts is a fairly common usage scenario) 
- never connect to your unrelated web server again (hyperbole, I have backups of my keys)

**Expected behaviour**
- prefill the value from id_rsa.pub in the field 
and 
- before running the keygen command, **check** if a keypair exists at the location and **don't overwrite** it
   - maybe prompt the user that the key already exists and ask for a new file path 
   - or just error out, idk, just leave my existing keys alone

**Desktop (please complete the following information):** 
- OS: macOS 14.1.1 on apple silicon
- Imager v1.8.4

**The name of the OS you are trying to write**
- Raspberry Pi OS (64-bit) 

**Are you using OS Customisation?**
Yes, this problem occurs during the OS Customization step.

tdewey-rpi, 2024-01-10:
Yikes! This is _absolutely_ not intended. Transferring to work on this one now.

tdewey-rpi, 2024-01-10:
Unable to reproduce this issue.

On 1.8.4, I see:

- my id_rsa file is populated automatically
- In the code, the generation function is wrapped in a file existence check

tdewey-rpi, 2024-01-10:
Inspection of the codebase confirms we will not overwrite an id_rsa file that exists at the point we come to write it:

https://github.com/raspberrypi/rpi-imager/blob/6f6a90adbb88c135534d5f20cc2a10f167ea43a3/src/imagewriter.cpp#L1106

The only way I can see your issue to have happened is if there was a TOC/TOU around the file existence check - but your described steps do not state a situation where that would occur.

tdewey-rpi, 2024-01-10:
@vovinhd Please confirm how you installed rpi-imager, along with confirmation the reproduction steps are exactly what you did.

vovinhd, 2024-01-10:
I've downloaded rpi-imager from https://www.raspberrypi.com/software/ (as anyone would, I suspect), earlier today. 
[image]
^see the 'where from' and created time 
here's a hash of the file, if that helps:
```
$ md5 imager_1.8.4.dmg
MD5 (imager_1.8.4.dmg) = 69d6dc02ebe166b241b665eeb1b00fb9
```

I don't see how I could be more granular in my reproduction steps, and that is what happened.
Before running the imager I had a id_rsa/id_rsa.pub keypair that I used, afterwards it was gone. 

I don't see how that could have happened. From looking at the source file you just linked it looks covered, but I'm not a Qt expert. 
Maybe it is a weird permission thing with macOS? But if it could not stat the tile should not have been able to overwrite the file either

tdewey-rpi, 2024-01-10:
@vovinhd I think we're in the same place, then: utter confusion about how this occurred, and looking at QFile in a very suspicious way.

Thanks for confirming provenance & steps. While I'd like to take those on first look, I've had too many reports where both turned out to be quite different when challenged.

Did you have id_rsa symlinked, or similar?

vovinhd, 2024-01-10:
I think I figured it out, and it is (at least partially) my fault >_<

When setting up my machine I only copied over id_rsa from my configs dir to .ssh, and not both id_rsa and id_rsa.pub (which means my initial report was inaccurate in that regard. Sorry.)

I recreated those starting conditions but I can't reproduce the error again, it now fails silently (as it should, I looked very suspiciously at line 1108, but it checks out) 

I guess blame it on cosmic rays?

tdewey-rpi, 2024-01-10:
Thanks for following up, @vovinhd. Explanation makes perfect sense.

I'll look at improvements around the key generation flow for 2.0, which we're now in planning for. If you've got any feedback about how you'd expect that to work, I'd love to hear it.

lurch, 2024-01-10:
Can you add `QCosmicRayProtection()` ? :laughing:

tdewey-rpi, 2024-01-22:
Closing as stale - no additional feedback, so we'll go ahead with planning.

closed 2024-01-22 by tdewey-rpi