raspberrypi/rpi-imager #784: [FEATURE] Import SSH Keys from GitHub username

nelsonjchen, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

The device I'm flashing on may not have a copy of all my public keys. Or it might have just one of my public keys and I have other workstations. But I do know my GitHub username and that does have all or most keys from most of my workstations...

https://github.com/nelsonjchen.keys

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Be able to enter in a GitHub username and import all public SSH keys from GitHub as an option for pre-filling SSH keys when flashing.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Everything else is work. I'm lazy. 

**Additional context**
Add any other context or screenshots about the feature request here.

I've seen this in Ubuntu Server's installer and comma.ai openpilot's SSH setup UI. In both of those cases, "Import from GitHub" was extremely convenient. I suspect there could be other examples out there as well.

lurch, 2024-01-15:
Duplicate of #374 ?

nelsonjchen, 2024-01-15:
Wow! I was searching issues quite furiously and never would have guessed it was already at languishing PR stage.

tdewey-rpi, 2024-01-15:
Tagging this for 2.0. This is a pretty powerful feature, and it will require some additional UI space to look natural inside Raspberry Pi Imager.

tdewey-rpi, 2026-10-10:
A version of this now ships in the Network Installer - I've tried it, and it's ludicrously effective. I'll consider promoting it to the desktop for 2.1.