raspberrypi/rpi-imager #787: Username and/or Password get transfered faulty

JoniDev25, 2024-01-19:
When I Try Installing Any Ubuntu Server LTS On My Raspberry PI 4b And Start My Raspberry PI, It Says That The Username and/or Password Is Incorrect Upon Login. I Tried Reimaging My SD Card About 17 Times But The Problem Doesnt Get Fixed. I Also Think That The Settings-Prompt When Trying To Install An Operating System In The Raspberry PI Imager Might Be A Bit Confusing And It Might Be Better To Be Able To Set It Up Before Selecting A Board And An OS.

Reproduce:
1) Go Into The Raspberry PI Imager And Flash Any Ubuntu Server LTS On The Raspberry PI 4
2) Press Change Settings Upon Asked
3) Create A Custom Username And Password
4) Press Any Button Except Edit Settings After That And Flash It (I Tried Every Button)
5) Login With Your Custom Username And Password In SSH Or On The PI Itself

Expected:
I Get Logged In Without Any Warnings Or Problems

It Just Says Login Failed Like As If I Would Type Wrong Username/Password Into The Terminal. I Already Checked If I Typed It Correctly Multiple Times And It Was. I Even Changed The Password In The Imager x5 Times.

Desktop:
Windows 11

Programs:
Raspberry PI Imager v1.8.4

Images:
Ubuntu Server LTS 22.04.03
Ubuntu Server LTS 20.04.05

The OS Is Cuszomized With WiFi, Username, Password And SSH. It Worked On Any Other Images With These Settings (Before The New Update. In The New Update, I Only Tested Ubuntu Server LTS But Ubuntu Server LTS Worked As Well In The Old Version Of Raspberry PI Imager)

tdewey-rpi, 2024-01-22:
Thanks for the report, @JoniDev25 

UI changes are deferred to v2.0 unless they are an active source of bugs. 

Can you provide precise detail on what you did between these two steps, please:

> Create A Custom Username And Password
> Press Any Button Except Edit Settings After That And Flash It (I Tried Every Button)

Please also confirm if you're seeing this behaviour on other images, and if you see this failure with a password comprised entirely of standard English alphabet characters (no symbols, numbers).

JoniDev25, 2024-01-23:
Well, before The New Update came Out, The Raspberry Pi imager looked Different with The Options Button visible before flashing The Image. I Configured Everything Including Wi-Fi, Keyboard, Hostname, SSH and Language Settings Like I did before, But It Didn't Work Anymore. The Password and/or Username have been refused By The Raspberry Pi. I checked Multiple Times to see if I had written everything correctly by typing the password in the username field of The Ubuntu Server Terminal so It was Visible And the correct one. I did that In Every Installation I Made. 

Regarding Your Questions To The Buttons:
There Are 4 Buttons: Edit Settings, Flash without Settings, Yes and No. I flashed My Image With The Exact Same data every time, Using A different one of The 4 Buttons Every time. The Password and/or Username still always showed as "wrong" In The Ubuntu Server Terminal.

[image]
-> The 4 "Buttons"

To be specific: 
I flashed The Image with The Same Username and Password 4+ Times using one of The Buttons each Time until I went All of them Through. But I still had that same error appear each time.

maxnet, 2024-01-23:
Ubuntu Server 22.04 (64-bit) does work for me.

It is only the password you are having problems with, and it did apply other settings?
E.g. if you set a custom host name it does show that on the login screen like this?

[image]

And have you tried other operating systems like RPI OS?

JoniDev25, 2024-01-24:
Everything works except the username and password. I get the correct custom hostname and it connects to my Wi-Fi automatically. And no, I only tried Ubuntu Server so far and even if it only is for that specific installation, I still dont know what do change. I need that specific operating system. 

One Question:
Can you please give Me A step to step instruction On how you installed Ubuntu Server On your Raspberry PI? Maybe I’m going wrong somewhere.

lurch, 2024-01-24:
> One Question:
> Can you please give Me A step to step instruction On how you installed Ubuntu Server On your Raspberry PI? Maybe I’m going wrong somewhere.

That's the kind of user-support question that ought to be asked on https://forums.raspberrypi.com/ not on GitHub.

maxnet, 2024-01-24:
>I need that specific operating system.

Could be.
But right now we do not know if it is an Ubuntu specific problem that may only occurs in very special circumstances, that others may not see.
Or something wrong with your Imager installation.
So still suggest you try RPI OS as well to narrow down the problem.

We had some very rare problems that occured only under very special circumstances before.
Like an imager crash, that only occured if the user was running on a Chinese MacOS version.

Or problems with advanced options that only occured when multiple options were set to a certain combination.
E.g. if it errors out setting the customization for something else,  e.g. keyboard layout, it may stop there, and may not get to setting the password in some circumstances, on some operating systems. Things like that.

JoniDev25, 2024-01-26:
Hi Again. I Just did some research and i found my problem. Apparently, the username "admin" is reserved by the system. Maybe i should have mentioned that i used that username. I just didnt know that it would have mattered. But thanks for yall help anyways!

JoniDev25, 2024-01-26:
But maybe it would still be a great feature if the imager warned people when the username is reserved and cant be used for a specific installation.

closed 2024-01-26 by JoniDev25

maxnet, 2024-01-26:
>But maybe it would still be a great feature if the imager warned people when the username is reserved

Practical problem is that we offer a dozen different operating system images made by third-parties, and Imager currently does not have any information about which user names are already in-use on each.