raspberrypi/rpi-imager #788: [BUG] Error removing exsting partitions

AnmolrPunj, 2024-01-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Include steps to allow us to reproduce the error. Include your host OS, and the version of Raspberry Pi Imager.
Wubdiws 10 version 1.8.4

**Expected behaviour**
A clear and concise description of what you expected to happen.
Should format the sd card
**Screenshots or video**
If applicable, add screenshots or a video to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. macOS 14, Windows 11, Ubuntu Server 22.04]

**The name of the OS you are trying to write**
Raspberry pi os legacy 32bit rpi os 64bit rpi os legacy 64 bit
**Are you using OS Customisation?**
No
**Additional context**
Add any other context about the problem here.
Image:
[image]

tdewey-rpi, 2024-02-23:
Are you writing to an SD card? If so, have you nudged the write protect switch?

AnmolrPunj, 2024-02-23:
i fixed it by puting the sd card in another computer now it works on my computer as well

lurch, 2024-02-24:
> now it works on my computer as well

So I guess this issue can be closed?

tdewey-rpi, 2024-02-28:
Closing as 'Won't Fix' - there's no sign of an issue to be addressed in the Raspberry Pi Imager codebase.

closed 2024-02-28 by tdewey-rpi