raspberrypi/rpi-imager #793: [BUG] Cannot login on first boot after imaging - "No directory, logging in with HOME=/ This account is currently not available."

0nobody0, 2024-01-27:
**Describe the bug**
on first boot after successfully verified imaging with custom user set, i cannot login. After attempting to login a black screen with a cursor appears and then the screen instantly switches back to the login screen. Switching to tty1 i see the message:
```
No directory, logging in with HOME=/
This account is currently not available.
```

**To Reproduce**
Using Raspberry Pi Imager v1.8.5 on Windows 11 x64 to image a microsd card with 'Raspberry Pi OS with desktop' (Release date: December 5th 2023, System: 64-bit, Kernel version: 6.1, Debian version: 12 (bookworm). I specified a custom username and password in the settings before imaging.

Boot and try to login.

**Expected behaviour**
User would have home directory set, be able to login

**Screenshots or video**
n/a

**Desktop (please complete the following information):**
 - OS: Windows 11 x64

**The name of the OS you are trying to write**
 Raspberry Pi OS with desktop - Release date: December 5th 2023, System: 64-bit, Kernel version: 6.1, Debian version: 12 (bookworm)

**Are you using OS Customisation?**
Yes

**Additional context**
thank you

maxnet, 2024-01-27:
Which username did you specify?

As "nobody" is usually a special system user, that is indeed not supposed to be able to login.

0nobody0, 2024-01-27:
yes, it was nobody. i forgot about that, thank you. will close this and reopen if the issue repeats for a different username.

closed 2024-01-27 by 0nobody0