raspberrypi/rpi-imager #794: [BUG] Unable to run on Ubuntu 22.04 LTS with default install

davija, 2024-01-28:
**Describe the bug**
Recent Ubuntu 22.04.3 LTS install.

**To Reproduce**
* Install current Ubuntu 22.04
* run `sudo apt install rpi-imager`
* run `rpi-imager`

**Expected behaviour**
Application starts up and I can use it.

**Screenshots or video**
[image]

In the console:
[image]

**Desktop (please complete the following information):**
Ubuntu Desktop 22.04.3 LTS

**The name of the OS you are trying to write**
Can't get that far

**Are you using OS Customisation?**
Can't get that far

**Additional context**
I have tried with both qt5 & qt6 installed and always get the same result.  Looking around on the internet and this is a VERY common problem and it appears that this may be linked against an incorrect OpenSSL version.

lurch, 2024-01-28:
Your screenshot shows that you're using Raspberry Pi Imager v1.7.2, but v1.8.5 is available from https://github.com/raspberrypi/rpi-imager/releases

tdewey-rpi, 2024-01-29:
Hi @davija,

Unfortunately we do not maintain the package in the Ubuntu repositories, and as such installations via this mechanism are unsupported if we believe the problem is related to packaging.

In this case, it appears Qt has attempted to open OpenSSL functions and failed to do so, which would likely produce the failure you've reported. There are numerous reasons this may have happened - but the most likely by far is a failure of your package manager's dependency resolution mechanism, which again on the balance of probability would have been caused by rpi-imager being incorrectly packaged.

I'd encourage you to install rpi-imager directly from this repository where possible, and install OpenSSL from your OS package manager:

`sudo apt install -y openssl`

Closing, as on the balance of probability this is not an issue with Raspberry Pi Imager, this is an issue with Ubuntu 22.04's package repository.

closed 2024-01-29 by tdewey-rpi

lurch, 2024-01-29:
ping @waveform80 in case he's able to fix the Ubuntu packaging (?).

maxnet, 2024-01-29:
>Unfortunately we do not maintain the package in the Ubuntu repositories

While that is true, I have a feeling our own package is unlikely to work either in cases like this. 

>I have tried with both qt5 & qt6 installed

And how did you install those exactly?

If you are using Ubuntu's own Qt packages, everything should be fine.
libqt5network5 pulls in the right libssl3

However if you got your Qt elsewhere. E.g. from the Qt site.
Things can get more interesting. ;)
Especially if the version you got isn't meant for this Ubuntu version, but you installed it at some point in the past, and upgraded your Ubuntu later...

waveform80, 2024-02-02:
Sorry for the delay replying -- the version of rpi-imager in jammy is indeed 1.7.2 and thus rather old, but it does still work for me on both PC and Pi which, as noted, suggests the issue is something "unusual" in the Qt installation.

While I can't bump the version in jammy (because it still works), the snap should be up to date (ish -- just realized I need to bump it to 1.8.5) and *should* (I think?) work even in the presence of a broken Qt installation (though only because the snap bundles its own Qt, making it hilariously large, because the Qt content snap is only built for amd64/arm64 and I need to include at least armhf as well for rpi-imager).

davija, 2024-02-08:
Sorry for the delayed response...

In addition to the version in the ubuntu repository, I also downloaded the most recent version (1.8.5) and got very similar errors with that one.  I did not get the same error downloading OS list message, however in the console I got similar messages and none of the dropdowns had any content and I was not able to do anything with it.

I did try the snap and that did work.  Not sure why using the latest version installed manually via the .deb package wasn't working as it should have been using the newer Qt version that was correctly linked with the newer openssl version.