raspberrypi/rpi-imager #795: [BUG] Embedded: Pi 4: Imager doesn't see my USB SATA drive

andrum993, 2024-01-29:
**Describe the bug**
Pi 4B 8GB. Running Network Install (embedded Imager) to do a reinstall on my USB-connected SATA drive. Bridge is ASM1153 to a Kingston A400 series 2.5" SATA drive.

**To Reproduce**
1. Boot Pi 4 with no boot media present and Ethernet connected.
2. Hold shift to enter Network Install.
3. Plug in USB SATA drive.
4. Click 'Choose Storage' button in Imager.

or

1. Boot Pi 4 with USB SATA drive present but no Pi boot files present.
2. Hold shift to enter Network Install.
3. Click 'Choose Storage' button in Imager.

**Expected behaviour**
Drive shows up in Imager's 'Storage' dialog. Can image drive OK.

**Actual behaviour**
Imager's 'Storage' dialog is empty. Cannot image drive using embedded Imager.

maxnet, 2024-01-29:
Suspect the problem is that it is an UAS protocol bridge claiming the drive is non-removable.

Can you try if booting the following files from a FAT32 formatted SD card works better?

[embedded-20240129.zip](https://www.maxnet.eu/tmp/embedded/embedded-20240129.zip)

andrum993, 2024-01-29:
It was working previously with the embedded Imager, so something's changed.
Is there a way of getting to a shell within the Network Install environment
to have a look? Also this was working fine on Pi 5 with this exact same
drive last week.

maxnet, 2024-01-29:
>It was working previously with the embedded Imager, so something's changed.

Probably the lshw version, which we outsource disk enumeration to.
Arch users also complained about a newer versions indicating drives were non-removable that were removable before.

>Is there a way of getting to a shell within the Network Install environment to have a look? 

Yes, put blockingshell in cmdline.txt
Or set console to serial, and use debugshell instead

There are more problems with UAS drives.
My JMicron based one gives writes errors half way through Imager writing the image.

andrum993, 2024-01-29:
Thanks - the files in the zip file worked - drive shows up and I can successfully image the drive 👍

tdewey-rpi, 2024-02-07:
@andrum993 Just confirming - did you later encounter the issues that @maxnet suggested may occur?

andrum993, 2024-02-07:
> @andrum993 Just confirming - did you later encounter the issues that @maxnet suggested may occur?

No. This is a known working setup - the only part that wasn't working was embedded Imager, until maxnet fixed it.

I was waiting until the fix showed up in the Network Install version of embedded Imager before I closed this issue. I've just checked, and the fix has not yet made it to Network Install, so I would propose to keep this issue open.

tdewey-rpi, 2024-02-07:
Agreed - I'll hold it open until we've actually got the network install version out.

andrum993, 2024-04-18:
Just a heads up that Network Install is now enabled in the default bootloader EEPROM on Pi 5, but this bug prevents its use on any drive that Linux uses UAS on. Hopefully embedded Imager will get updated soon🤞

andrum993, 2024-05-07:
Hi @tdewey-rpi. Do you have an ETA for when this fix will find its way into Network Install - starting to get a bit annoying.

tdewey-rpi, 2024-05-08:
> Hi @tdewey-rpi. Do you have an ETA for when this fix will find its way into Network Install - starting to get a bit annoying.

I've not had the bandwidth to look at this yet, but will take a look this week as it may address a range of concerns with the embedded Imager.

JackDean, 2024-06-08:
I can confirm that this is still broken for PI 4.

andrum993, 2024-06-20:
This is now working for me with the current Network Install image from `<official server>/net_install/boot.img` with file datetime `2024-06-20 16:39` on a Pi 4B 8GB Rev 1.4. Works both with target drive attached while the Pi boots (no valid Pi boot files on it) and if I leave it unplugged, then attach it once embedded Imager has loaded.

Many thanks! 👍😄

PR #884 (2024-06-25): BCM2712d0 support
Uprate the buildroot distribution we use for the embedded imager, selecting a new kernel, firmware and mesa in the process.

Tested on BCM2712d0.

Resolves #795
closed 2024-06-20 by andrum993