raspberrypi/rpi-imager #797: [BUG] Embedded Imager on Pi5 runs fan at full-tilt

lurch, 2024-01-30:
**Describe the bug**
When running Embedded Imager (network OS installer) on a Raspberry Pi 5 with an Active Cooler attached, the fan is running full-speed the whole time, which is slightly annoying. I'm sure that writing an OS image to an SD-card can't be very CPU-intensive!

**To Reproduce**
Connect up power to a Pi 5 with a blank micro SD card inserted, connected to a USB keyboard, USB mouse, HDMI monitor and Ethernet cable. Follow the on-screen instructions to load the embedded version of RPi Imager.

**Expected behaviour**
The fan should turn off if the CPU is under low-load (just like it does in Raspberry Pi OS).

**Screenshots or video**
If applicable, add screenshots or a video to help explain your problem.

**Desktop (please complete the following information):**
 - OS: None!

**The name of the OS you are trying to write**
Raspberry Pi OS 64-bit

**Are you using OS Customisation?**
Yes, but that doesn't matter

**Additional context**
Add any other context about the problem here.

timg236, 2024-01-31:
It’s possibly missing the fan dt overlay.

tdewey-rpi, 2024-07-31:
I believe this should now be resolved - @lurch can you confirm?