raspberrypi/rpi-imager #798: [BUG] imager 1.8.5 stuck on "choose storage"

tomasbjerre, 2024-01-30:
**Describe the bug**
When I click on "choose storage" the GUI freezes.

**To Reproduce**
* Install imager 1.8.5
* Open with `sudo rpi-imager` or `rpi-imager`
* Select Raspberry Pi 5
* Select Raspberry PI OS 64 bit
* Click "Choose storage"

GUI is now stuck 

**Expected behaviour**
Not stuck.

**Screenshots or video**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04

**The name of the OS you are trying to write**
Raspberry pi os 64 bit

**Are you using OS Customisation?**
No

**Additional context**
I see this output in the console when I start Imager:
```sh
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```

After a while, when it is stuck, I see this output in the console:
```
qrc:/main.qml:795:9: QML QQuickItem: Binding loop detected for property "height"
```

tdewey-rpi, 2024-01-30:
Thanks for the report, @tomasbjerre

When you say "stuck"; are you able to close the Storage window?

Neither of the messages you've highlighted are fatal, though not being able to determine the XDG runtime directory is a strange failure.

I'll take a look at reproducing this later in the week - but more details on the storage devices you have connected and the precise behaviour of the application in this state would be great.

maxnet, 2024-01-30:
Running "lsblk" in a terminal does not hang, and does display devices that have an "1" in the "RM" (removable) column?

tomasbjerre, 2024-01-30:
Yes I am able to close it.

Actually I found out that the sdcard reader was not connected. So there were no sd card or any USB sticks plugged in.

I solved this by using a working sdcard reader and the GUI worked just fine with that.

tomasbjerre, 2024-01-30:
Here is `lsblk`:

```sh
bjerre@bjerre-ubuntu:~$ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0    7:0    0     4K  1 loop /snap/bare/5
loop1    7:1    0 158,9M  1 loop /snap/chromium/2734
loop2    7:2    0 159,6M  1 loop /snap/chromium/2738
loop3    7:3    0 105,8M  1 loop /snap/core/16091
loop4    7:4    0 105,8M  1 loop /snap/core/16202
loop5    7:5    0  55,7M  1 loop /snap/core18/2796
loop6    7:6    0  55,7M  1 loop /snap/core18/2812
loop7    7:7    0  63,5M  1 loop /snap/core20/2015
loop8    7:8    0  63,9M  1 loop /snap/core20/2105
loop9    7:9    0  74,1M  1 loop /snap/core22/1033
loop10   7:10   0  73,9M  1 loop /snap/core22/864
loop11   7:11   0  66,5M  1 loop /snap/cups/1024
loop12   7:12   0  66,6M  1 loop /snap/cups/980
loop13   7:13   0 262,1M  1 loop /snap/firefox/3728
loop14   7:14   0 240,3M  1 loop /snap/firefox/3290
loop15   7:15   0 140,7M  1 loop /snap/gnome-3-26-1604/104
loop16   7:16   0 140,7M  1 loop /snap/gnome-3-26-1604/111
loop17   7:17   0 164,8M  1 loop /snap/gnome-3-28-1804/194
loop18   7:18   0 164,8M  1 loop /snap/gnome-3-28-1804/198
loop19   7:19   0 218,4M  1 loop /snap/gnome-3-34-1804/90
loop20   7:20   0 218,4M  1 loop /snap/gnome-3-34-1804/93
loop21   7:21   0 349,7M  1 loop /snap/gnome-3-38-2004/140
loop22   7:22   0 349,7M  1 loop /snap/gnome-3-38-2004/143
loop23   7:23   0 496,9M  1 loop /snap/gnome-42-2204/132
loop24   7:24   0   497M  1 loop /snap/gnome-42-2204/141
loop25   7:25   0   1,5M  1 loop /snap/gnome-system-monitor/184
loop26   7:26   0   1,5M  1 loop /snap/gnome-system-monitor/186
loop27   7:27   0  81,3M  1 loop /snap/gtk-common-themes/1534
loop28   7:28   0  91,7M  1 loop /snap/gtk-common-themes/1535
loop29   7:29   0 842,4M  1 loop /snap/intellij-idea-community/475
loop30   7:30   0  37,1M  1 loop /snap/hunspell-dictionaries-1-7-2004/2
loop31   7:31   0 852,8M  1 loop /snap/intellij-idea-community/478
loop32   7:32   0  45,9M  1 loop /snap/snap-store/638
loop33   7:33   0  12,3M  1 loop /snap/snap-store/959
loop34   7:34   0   428K  1 loop /snap/snapd-desktop-integration/57
loop35   7:35   0   452K  1 loop /snap/snapd-desktop-integration/83
loop36   7:36   0 320,4M  1 loop /snap/vlc/3078
loop37   7:37   0 321,1M  1 loop /snap/vlc/3721
sda      8:0    0 476,9G  0 disk 
├─sda1   8:1    0   260M  0 part /boot/efi
├─sda2   8:2    0    16M  0 part 
├─sda3   8:3    0 258,2G  0 part 
├─sda4   8:4    0   499M  0 part 
└─sda5   8:5    0 217,9G  0 part /var/snap/firefox/common/host-hunspell
                                 /
```

I think it would be nice if the GUI can show some information about the state, perhaps "no storage detected" or similar.

tdewey-rpi, 2024-01-31:
Thanks for the feedback, @tomasbjerre

We'll look at your suggested UX improvement for 2.0.

Closing, as this was a HW configuration problem.

commit fba63d01 (2024-02-05): Add a message when there are no storage devices. Fixes #798
closed 2024-01-31 by tdewey-rpi