raspberrypi/rpi-imager #804: [BUG]

jonnytest1, 2024-02-06:
**Describe the bug**
when trying to install "generic oeprating system" > "ubuntu" > "ubuntu dekstop" there is not popup about using custom settings

**To Reproduce**
(1.8.5)
trying to install "generic operating system" > "ubuntu" > "ubuntu dekstop"

**Expected behaviour**
i get a popup asking about wifi credentials and username

**Desktop (please complete the following information):**
 - Windows 11

**The name of the OS you are trying to write**
ubuntu dekstop
**Are you using OS Customisation?**
trying to

lurch, 2024-02-06:
The popup now appears as part of the flashing process, rather than being a separate button in the GUI. See the example walkthrough at https://www.raspberrypi.com/documentation/computers/getting-started.html#raspberry-pi-imager

tdewey-rpi, 2024-02-06:
Closing as this is an intended behaviour.

Canonical appear to have disabled customisation support in their JSON file. Not an issue with rpi-imager.

closed 2024-02-06 by tdewey-rpi

maxnet, 2024-02-06:
>Canonical appear to have disabled customisation support in their JSON file. 

They never supported customization on Ubuntu Desktop.
Only Ubuntu server comes with Cloudinit.