raspberrypi/rpi-imager #808: [BUG] Ubuntu Server 23.10 (64 bit) Username and Password not Working

Clivern, 2024-02-06:
**Describe the bug**
Using advanced settings to set username and password for Provided Ubuntu Server 23.10 (64 bit not working while it does for Ubuntu Server 22.04.

**To Reproduce**
Using advanced settings to set username and password for Provided Ubuntu Server 23.10.

**Expected behaviour**
Username and password should work

**Screenshots or video**
If applicable, add screenshots or a video to help explain your problem.

**Desktop (please complete the following information):**
 - Ubuntu Server 23.10 (64 bit)

**The name of the OS you are trying to write:** Ubuntu Server 23.10 (64 bit)

**Are you using OS Customisation?** Yes

lurch, 2024-02-07:
ping @waveform80

tdewey-rpi, 2024-02-07:
@Clivern Could you please tell us what version of rpi-imager you're using? Are you also running Imager on Ubuntu Server 23.10?

Clivern, 2024-02-07:
> @Clivern Could you please tell us what version of rpi-imager you're using? Are you also running Imager on Ubuntu Server 23.10?

I did upgrade to latest rpi Imager v1.8.5 and it works fine now so closing the issue. Thanks @tdewey-rpi

closed 2024-02-07 by Clivern