raspberrypi/rpi-imager #811: [FEATURE] insert bookworm 32/64bit entry by selecting pi0w2

tigernero79, 2024-02-07:
I would like it if even by selecting p0w2 in the list of raspberry devices it not only carries legacy images but also 32 and 64bit bookworm iso. simply because it supports it like pi3 pi4 and pi5. I know that I can flash the image by not selecting the device but it would be nice if the 32 and 64 bit bookworm image was shown even by choosing p0w2 among the devices since it is also arm64

lurch, 2024-02-07:
Duplicate of #730

tdewey-rpi, 2024-02-07:
Thanks for the report, @tigernero79

Unfortunately, as @lurch points out, we've had this request a few times - and #730 covers the rationale behind why it's not presented today, and what I believe are the issues that you might face if you go down that path.

Closing as duplicate.

closed 2024-02-07 by tdewey-rpi