raspberrypi/rpi-imager #815: i got this verifying error every time after installing the OS image on SD card

ghiathkamel, 2024-02-08:
i got this verifying error every time after installing the OS image on SD card
please check the photo
[image]
**Desktop (please complete the following information):**
 Windows 11 

**The name of the OS you are trying to write**
Any RPI4 OS
**Are you using OS Customisation?**
No

lurch, 2024-02-08:
@ghiathkamel Creating a single new issue is enough. **Please** don't spam multiple old issues too.

tdewey-rpi, 2024-02-08:
Please confirm your card can be verified correctly - can you post a screenshot of the result of running H2testw against the storage device?

H2testw is available from https://www.heise.de/download/product/h2testw-50539/download and includes both German and English translations.

ghiathkamel, 2024-02-10:
> Please confirm your card can be verified correctly - can you post a screenshot of the result of running H2testw against the storage device?
> 
> H2testw is available from https://www.heise.de/download/product/h2testw-50539/download and includes both German and English translations.

you are right, the SD card damaged, not the rpi imager :)

closed 2024-02-10 by ghiathkamel