raspberrypi/rpi-imager #817: Storage > blank window⁉️

jurisuk, 2024-02-09:
After clicking the Storage button, a blank window opens and remains blank (regardless of the OS image).

I have tried this with v1.8.5, 1.8.1 and 1.7.5, to no avail.

Using macOS 14.4.

[image]

jurisuk, 2024-02-09:
Next, I am trying to install (compile) from source.
The first error I get is:
```
-- Performing Test atomicbuiltin - Success
CMake Error at CMakeLists.txt:111 (message):
  Missing suitable Qt library (must be at least version 5.14)

-- Configuring incomplete, errors occurred!
make: *** No targets specified and no makefile found.  Stop.
```
Hence, I will now look into upgrading my Qt. Wish me luck.

jurisuk, 2024-02-09:
I have installed (upgraded) Qt to v6 by using homebrew:
```
brew install qt
```
After this, the build from source progressed much further, but died due to errors:
```
[warnings omitted]
[ 88%] Automatic RCC for translations.qrc
[ 88%] Automatic RCC for qml.qrc
[ 88%] Building CXX object CMakeFiles/rpi-imager.dir/rpi-imager_autogen/mocs_compilation.cpp.o
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/mocs_compilation.cpp:3:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/moc_devicewrapper.cpp:9:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/../../../src/devicewrapper.h:11:
In file included from /opt/homebrew/lib/QtCore.framework/Headers/QFile:1:
/opt/homebrew/lib/QtCore.framework/Headers/qfile.h:59:58: error: 'path' is unavailable: introduced in macOS 10.15
inline QString fromFilesystemPath(const std::filesystem::path &path)
                                                         ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/include/c++/v1/__filesystem/path.h:442:24: note: 'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/mocs_compilation.cpp:3:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/moc_devicewrapper.cpp:9:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/../../../src/devicewrapper.h:11:
In file included from /opt/homebrew/lib/QtCore.framework/Headers/QFile:1:
/opt/homebrew/lib/QtCore.framework/Headers/qfile.h:64:40: error: 'native' is unavailable: introduced in macOS 10.15
    return QString::fromStdString(path.native());
                                       ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/include/c++/v1/__filesystem/path.h:775:22: note: 'native' has been explicitly marked unavailable here
  const string_type& native() const noexcept { return __pn_; }
                     ^
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/mocs_compilation.cpp:3:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/moc_devicewrapper.cpp:9:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/../../../src/devicewrapper.h:11:
In file included from /opt/homebrew/lib/QtCore.framework/Headers/QFile:1:
/opt/homebrew/lib/QtCore.framework/Headers/qfile.h:68:25: error: 'path' is unavailable: introduced in macOS 10.15
inline std::filesystem::path toFilesystemPath(const QString &path)
                        ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/include/c++/v1/__filesystem/path.h:442:24: note: 'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/mocs_compilation.cpp:3:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/moc_devicewrapper.cpp:9:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/../../../src/devicewrapper.h:11:
In file included from /opt/homebrew/lib/QtCore.framework/Headers/QFile:1:
/opt/homebrew/lib/QtCore.framework/Headers/qfile.h:70:29: error: 'path' is unavailable: introduced in macOS 10.15
    return std::filesystem::path(reinterpret_cast<const char16_t *>(path.cbegin()),
                            ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/include/c++/v1/__filesystem/path.h:442:24: note: 'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/mocs_compilation.cpp:3:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/moc_devicewrapper.cpp:9:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/../../../src/devicewrapper.h:11:
In file included from /opt/homebrew/lib/QtCore.framework/Headers/QFile:1:
/opt/homebrew/lib/QtCore.framework/Headers/qfile.h:70:12: error: 'path<const char16_t *>' is unavailable: introduced in macOS 10.15
    return std::filesystem::path(reinterpret_cast<const char16_t *>(path.cbegin()),
           ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/include/c++/v1/__filesystem/path.h:488:3: note: 'path<const char16_t *>' has been explicitly marked unavailable here
  path(_InputIt __first, _InputIt __last, format = format::auto_format) {
  ^
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/mocs_compilation.cpp:3:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/moc_devicewrapper.cpp:9:
In file included from /Users/peter/rpi-imager/build/rpi-imager_autogen/EWIEGA46WW/../../../src/devicewrapper.h:11:
In file included from /opt/homebrew/lib/QtCore.framework/Headers/QFile:1:
/opt/homebrew/lib/QtCore.framework/Headers/qfile.h:70:12: error: 'path' is unavailable: introduced in macOS 10.15
    return std::filesystem::path(reinterpret_cast<const char16_t *>(path.cbegin()),
           ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/include/c++/v1/__filesystem/path.h:488:3: note: 'path' has been explicitly marked unavailable here
  path(_InputIt __first, _InputIt __last, format = format::auto_format) {
  ^
[... 132 lines cut ...]
[... cut ...]
    std::filesystem::path filesystemSymLinkTarget() const
                     ^
/Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/include/c++/v1/__filesystem/path.h:442:24: note: 'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/rpi-imager.dir/rpi-imager_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/rpi-imager.dir/all] Error 2
make: *** [all] Error 2
```

jurisuk, 2024-02-09:
Can't wait, so I installed Balena Etcher and this app works fine!

tdewey-rpi, 2024-02-09:
Did you have a removable storage device connected to your Mac at the time you took the screenshot?

jurisuk, 2024-02-09:
I had a cm4 connected whilst doing the above

tdewey-rpi, 2024-02-09:
And had you booted the CM4 into MSD, or was it just connected?

jurisuk, 2024-02-09:
I do not understand.

tdewey-rpi, 2024-02-09:
Raspberry Pi Imager will only flash to removable storage devices recognised by the host OS - in this case, macOS. Unless the CM4 device was configured to expose it's storage as a Mass Storage Device, this behaviour is entirely as expected - because you've not actually provided a storage device for Raspberry Pi Imager to flash the device OS to.

Your response suggests that you did not follow the Mass Storage Device (MSD) guidance, and so I'm minded to close this issue as "Won't fix" - because Imager is behaving as expected.

jurisuk, 2024-02-09:
Fine. But as I've said, I've used rpi-imager before many, many times (on my earlier macOS) and all worked fine. 
All the best!

jurisuk, 2024-02-09:
Well, it was actually two issues that I was reporting.

1. The problem with rpi-imager failing on macOS 14.4
2. The source code not compiling on macOS 14.4

tdewey-rpi, 2024-02-09:
Attempted to reproduce on macOS 14 using a mass storage device (a UUGear eMMC Module connected to an Anker USB-C microSD/SD card reader/writer). Storage devices are enumerated as expected:

[image]

A change is queued for the next release to further emphasise that there are no storage devices detected.

tdewey-rpi, 2024-02-09:
> Well, it was actually two issues that I was reporting.

In the future, please try to make sure there is a 1:1 mapping of Issue Reports to Issues - it simplifies the discussion and tracking.

> 2. The source code not compiling on macOS 14.4

Looks like there's a typo in the CMakeLists - the OSX_DEPLOYMENT target is set to 10.14 in the case of Qt6, which is unexpected because we mandate 10.15 in the release.

maxnet, 2024-02-09:
>Looks like there's a typo in the CMakeLists - the OSX_DEPLOYMENT target is set to 10.14 in the case of Qt6, which is 
>unexpected because we mandate 10.15 in the release.

There is not really anything in the Imager code itself demanding 10.15
Reason we mention 10.15 on the download page is because 10.14 does not trust the SSL certificate authority used by downloads.raspberrypi.com 

You should use Qt 6.4 from the Qt website though and not homebrew.
Homebrew downloads latest Qt 6 subrelease, and latest 6 subrelease only supports latest OSes...